wled/WLED #1627: Build errors in PlatformIO

DIY3D, 2021-01-05:
First of all Best wishes for 2021 and a big thanks for creating this project!

I try to compile WLED in PlatformIO but the build ffailed with a few errors, below my terminal output:

`                 from wled00\cfg.cpp:1:
C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/bsp/include/nm_bsp.h:55:0: warning: "CONST" redefined [enabled by default]
 #define CONST const
 ^
In file included from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31:0,
                 from wled00\wled.h:53,
                 from wled00\cfg.cpp:1:
C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/IPAddress.h:42:0: note: this is the location of the previous definition
 #define CONST /* nothing: lwIP-v1 does not use const */
 ^
In file included from wled00\wled.h:75:0,
                 from wled00\alexa.cpp:1:
wled00\my_config.h:17:2: warning: #warning **** my_config.h: Settings from this file are honored **** [-Wcpp]
 #warning **** my_config.h: Settings from this file are honored ****
  ^
In file included from wled00\wled.h:75:0,
                 from wled00\cfg.cpp:1:
wled00\my_config.h:17:2: warning: #warning **** my_config.h: Settings from this file are honored **** [-Wcpp]
 #warning **** my_config.h: Settings from this file are honored ****
  ^
In file included from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/common/include/nm_common.h:45:0,
                 from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/socket/include/socket.h:60,
                 from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/WiFiUdp.h:24,
                 from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS_Legacy.h:47,
                 from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS.h:45,
                 from wled00\wled.h:54,
                 from wled00\blynk.cpp:1:
C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/bsp/include/nm_bsp.h:55:0: warning: "CONST" redefined [enabled by default]
 #define CONST const
 ^
In file included from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31:0,
                 from wled00\wled.h:53,
                 from wled00\blynk.cpp:1:
C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/IPAddress.h:42:0: note: this is the location of the previous definition
 #define CONST /* nothing: lwIP-v1 does not use const */
 ^
In file included from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/common/include/nm_common.h:45:0,
                 from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/socket/include/socket.h:60,
                 from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/WiFiUdp.h:24,
                 from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS_Legacy.h:47,
                 from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS.h:45,
                 from wled00\wled.h:54,
                 from wled00\button.cpp:1:
C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/bsp/include/nm_bsp.h:55:0: warning: "CONST" redefined [enabled by default]
 #define CONST const
 ^
In file included from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31:0,
                 from wled00\wled.h:53,
                 from wled00\button.cpp:1:
C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/IPAddress.h:42:0: note: this is the location of the previous definition
 #define CONST /* nothing: lwIP-v1 does not use const */
 ^
In file included from wled00\wled.h:75:0,
                 from wled00\blynk.cpp:1:
wled00\my_config.h:17:2: warning: #warning **** my_config.h: Settings from this file are honored **** [-Wcpp]
 #warning **** my_config.h: Settings from this file are honored ****
  ^
In file included from wled00\wled.h:75:0,
                 from wled00\button.cpp:1:
wled00\my_config.h:17:2: warning: #warning **** my_config.h: Settings from this file are honored **** [-Wcpp]
 #warning **** my_config.h: Settings from this file are honored ****
  ^
In file included from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/common/include/nm_common.h:45:0,
                 from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/socket/include/socket.h:60,
                 from C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/WiFiUdp.h:24,
                 from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS_Legacy.h:47,
[... 334 lines cut ...]
   virtual uint8_t beginMulticast(IPAddress, uint16_t);  // initialize, start listening on specified multicast IP address and port. Returns 1 if successful, 0 if there are no sockets available to use
                   ^
C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/WiFiUdp.h:47:19: note:   candidate expects 2 arguments, 3 provided
wled00\src/dependencies/espalexa/Espalexa.h:346:101: note: candidate is:
In file included from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS_Legacy.h:47:0,
                 from C:\Users\Marco\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src/ESP8266mDNS.h:45,
                 from wled00\wled.h:54,
                 from wled00\alexa.cpp:1:
C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/WiFiUdp.h:47:19: note: virtual uint8_t WiFiUDP::beginMulticast(IPAddress, uint16_t)
   virtual uint8_t beginMulticast(IPAddress, uint16_t);  // initialize, start listening on specified multicast IP address and port. Returns 1 if successful, 0 if there are no sockets available to use
                   ^
C:\Users\Marco\.platformio\lib\WiFi101_ID299\src/WiFiUdp.h:47:19: note:   candidate expects 2 arguments, 3 provided
*** [.pio\build\d1_mini\src\alexa.cpp.o] Error 1
*** [.pio\build\d1_mini\src\blynk.cpp.o] Error 1
============================================ [FAILED] Took 38.69 seconds ============================================

Environment    Status    Duration
-------------  --------  ------------
d1_mini        FAILED    00:00:38.688
======================================= 1 failed, 0 succeeded in 00:00:38.688 =======================================
The terminal process "C:\Users\Marco\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

Can you point me in the right direction?

Aircoookie, 2021-01-08:
Hi!
This is a strange problem :) It might have mixed things up with library downloads from a previous project. WLED does not use the Wifi101 library, which is referenced in the error log. You can try deleting the `.pio` folder in your WLED folder (not `pio` but `.pio`) and it should re-download the required libraries, which might fix the issue.

DIY3D, 2021-01-08:
Hi,
 I deleted the .pio folder but had the same errors. Then I tried to compile it on another laptop with success! Amazing code! Thanks again.

closed 2021-01-08 by DIY3D