wled/WLED #1629: D1 Mini dropping off the network

psinide, 2021-01-06:
The board drops off the network after sometime. The most its stayed on the network is about 5-6 hours. I've tried using the Disable WiFi sleep option, used boards from different brands & also used 3 different WS2812B LED strips. Either a reset, power cycle or even rebooting the Wifi router brings it back on the network

**To Reproduce**
I have reproduced this with the D1 Mini connected to the same power supply as the LED strip, connected to different power supplies, connecting only the D1 Mini to power and also same combinations when connecting to a computer. It has never lasted through out the night

Have also commented in issue #424 and used @gegu 's ping usermod, but face the same issue.

**WLED version**
 - Board: Wemos D1 Mini
 - Version: 0.11.1
 - Format: used a binary and also self-compiled firmware

**Additional context**
Flashed ESPhome with FastLED on the same boards and the setup is working for 3 days and counting now. Will be trying on a nodemcu board soon.

Thank you for your help!

Travisco29, 2021-01-07:
Having the same issue. reverting to the previous release fixed it

psinide, 2021-01-07:
> Having the same issue. reverting to the previous release fixed it

Which version? 0.11.0?

Travisco29, 2021-01-07:
yes that is what i reverted too

WoodyLetsCode, 2021-01-07:
How many LEDs do you have?

psinide, 2021-01-07:
60 LEDs on a 1m strip, powered with a 5v 1.8mA USB power adapter. Currently reverted back to 0.11.0 and no issues for 2 hours

Travisco29, 2021-01-07:
same thing was happening with one of my nodemcus, reverted and its fine now

psinide, 2021-01-08:
17 hours and counting. The first one to go through the night! Looking hopeful, will continue to test over the weekend and switch back to WLED once I'm convinced! Currently I only have the D1 Mini connected to power with no LEDs, will continue the tests with a strip connected and keep you posted.

psinide, 2021-01-09:
I got a full night with the LEDs connected, but it dropped off while i was away. It could be due to external factors. Going to test further and post results here.

Travisco29, 2021-01-09:
How are you powering the leds? Directly into the d1 or external power supply

psinide, 2021-01-09:
> How are you powering the leds? Directly into the d1 or external power supply
> […](#)

Directly into the D1. The D1 is connected to a 5v 2A USB adapter and I've limited the current draw and peak brightness via Wled to 1500mA

Travisco29, 2021-01-09:
I had issues with every version with that approach and switched to external

psinide, 2021-01-09:
> I had issues with every version with that approach and switched to external
> […](#)

My ESPHome LED strips are working with out any issues in the same setup. I'll try an external supply as well for testing, but this setup is not feasible for my use case. These LEDs are accent and ambient lights, extra wires will not make them aesthetic

psinide, 2021-01-09:
@Travisco29 are you using an external power supply (non-usb) to power the D1? If so, can you share your schematic?
I don't have a logic shifter, so testing this will have to wait till the component is delivered

psinide, 2021-01-11:
Current uptime - 1 day, 12 hours. No changes done with the connections. The strip is is still powered via the D1

psinide, 2021-01-12:
It was working well till last night. But now started dropping again since today morning

Aircoookie, 2021-01-13:
@psinide I'm very sorry, this must be frustrating. What router model are you using?
I haven't found the reason for worse WiFi reliability in WLED compared to Tasmota and ESPHome yet unfortunately. Definitely does not sound like a hardware issue though.

psinide, 2021-01-13:
@Aircoookie really appreciate what you've done here! I'm using a Netgear X8000 and the D1 is about 3 feet away with line of sight as well. I'm waiting for the logic shifter to be delivered to try power the Led strip and the D1 mini separately.

Doyle4, 2021-01-13:
Glad its not just me, 11.0. and 11.1, once SSD and Password is entered and rebooted, I have to reflash the firmware as it will not connect to my home router and no way of access the board its self using IP or AP even with it enabled. Thought it may of been a faulty board, but all of my ESP32's and D1 Mini Pros all will not connect.

flibble2000, 2021-01-15:
I'm also experiencing similar issues, since I discovered WLED with v 0.10.2. I have 4 d1 minis, each have a 12v power supply for the WS2815 and the D1 is powered by usb from a 12v stepdown converter. There is common ground between LED's and the D1s. The PSU's are all 12v 5amp output, and the stepdown has a 3amp max rating. Both run well within power specifications has there is no more than 120 LED's on a single strip.

All appear to stop communicating at what seems to be random times. All 4 do not stop at the same time, some will stay up for a couple of days then need resetting, others will be just a few hours. There is no pattern to which D1's stop, so one unit could stay up for a couple of days, but then stop repeatedly over a 24 hour period.

I have compiled by own with the Fixunreachablenetservices usermod, and tried various polling intervals in this from 10 to 60 seconds and it doesn't appear to make a difference. I have also turned off the sync options.

When they stop communicating, they all stay connected to wireless and I can get them working again by dropping them off the network from my wireless controller.

My wireless setup is a Fortinet-WLC with a number of AP's around the house. The IOT network is hidden (I have tried with it not being) and the D1's stick to the closest AP which is in the kitchen. They stick to this AP with signal strength ranging between 45-60db.

I have moved to ESPHome for now just to get some reliable lighting going on, they have been rock solid for the passed 2 weeks now. But would like to move back in the future and if there is any testing that needs doing I'm happy to help.

psinide, 2021-01-15:
I have also moved to ESPHome for now. I'm waiting for my nodeMCU & ESP32 boards to arrive, will test WLED on those. It seems that D1/mini suffers the most from this problem

SebastianBar, 2021-01-17:
> I have also moved to ESPHome for now. I'm waiting for my nodeMCU & ESP32 boards to arrive, will test WLED on those. It seems that D1/mini suffers the most from this problem

Three ESP32's are dropping after the latest updates as well, just for you to know (on v0.9 and v0.10 never had this issue AFAIK).

pk88888, 2021-01-20:
I have six custom boards with V0.11.1 and noticed the same behavior as others. All boards with esp32 are dropping connection after few hours or sometimes after few days.

jlegen, 2021-01-24:
I'm also hit by WLED UI (0.11.1 on D1 minis) getting unresponsive after some random time. And i also noticed that they're then still connected to Wifi. 
I am using a work around now, which eliminated the problem for me: a cron job is issuing a "ping" to WLED's IP every minute. Works for some weeks now... :-)

Travisco29, 2021-01-25:
That's a good idea but i am probably just going to try switching to esp32s
and see if that fixes it.

Doyle4, 2021-01-25:
I have to ask, how are you guys even getting them to connect to your router? 1 of mine did, the other 4 I have to re flash, soon as I press save after adding SSD and password, the device is useless as it wont connect to the router and no AP shows even when enabled, leading to be re-flashed.

psinide, 2021-01-27:
> I'm also hit by WLED UI (0.11.1 on D1 minis) getting unresponsive after some random time. And i also noticed that they're then still connected to Wifi.
> I am using a work around now, which eliminated [...]

Where is your cron job running? There is a usermod which pings the router from the WLED firmware, I was using it, and for a while it all worked well. The ESP was on the network for 3 days without a drop and then suddenly it started failing. I tried to re-flash it with the usermod, reduced the ping intervals to 30 & 10 seconds, but still it would drop off.

> I have to ask, how are you guys even getting them to connect to your router? 1 of mine did, the other 4 I have to re flash, soon as I press save after adding SSD and password, the device is useless as it wont connect to the router and no AP shows even when enabled, leading to be re-flashed.

I have faced no issues with the setup. After flashing, I can see the AP and then can configure my SSID and it connects to my network. The intermittent drops from the network are the only problem I'm facing.

jlegen, 2021-01-27:
> Where is your cron job running? There is a usermod which pings the router from the WLED firmware, I was using it, and for a while it all worked well. The ESP was on the network for 3 days without a [...]
i am using iobroker (also for WLED), so i have a 24/7 server running...

> I have faced no issues with the setup. After flashing, I can see the AP and then can configure my SSID and it connects to my network. The intermittent drops from the network are the only problem I'm facing.

same here, setup etc. worked like a charm; the problem with the UI/web drop almost feels like a "powersave feature", where somehow the wakeup doesn't work properly. At least it's probably not IDF/ESP8266-Core related, since Tasmota is based on that as well, and is running fine "for years" on those D1 minis...

Travisco29, 2021-01-27:
didn't help me but did you try the disable wifi sleep option in the wifi
settings?

jlegen, 2021-01-27:
> didn't help me but did you try the disable wifi sleep option in the wifi settings?
> […](#)

Yes, didn‘t change anything - which makes sense, because Wifi does not go sleep in this case, connection is still listed in router, and ping would also not work without Wifi...

WJCFerguson, 2021-01-30:
Same issue - loss of connectivity within hours, definitely less than 24 hours (sorry, not quantified).

DHCP lease is kept active at the router (my router gives 10 minute leases), so networking is alive at some level, but the web interface isn't reachable, nor does it respond to pings.

D1 mini ESP8266 clone, 144 WS2812B attached.

dredvard, 2021-01-30:
Exacly

Same issue - loss of connectivity although mine seems to last longer.
 
> DHCP lease is kept active at the router (my router gives 10 minute leases), so networking is alive at some level, but the web interface isn't reachable, nor does it respond to pings.
 
WiMOs D1 Mini 120 WS2812B attached.

Travisco29, 2021-02-02:
anyone tried esp32s and do they have the same issue?

pk88888, 2021-02-03:
Try this method:

1. Run Flash download tools (https://www.espressif.com/en/support/download/other-tools)
2. Connect to the module and delete the flash memory
3. Upload the new code using Visual Code according to the WLED instruction.

In my opinion, the problem is caused by wrongly uploading ready bin file.
It helped me and now the devices have been working properly for several days.

WJCFerguson, 2021-02-05:
I did a full flash erase, reflash, and verify flash all via `esptool.py`, and still suffer the same problem.  Perhaps at the weekend I'll

Urchin73, 2021-02-07:
I have 2 D1 mini (Aliexpress) running WLED 11.1. They are both at 50cm from wifi bridge. They are powered from the same power supply and led are powered directly from the power supply, not via the D1 minis.
1 mini act as a 'master', sending while the other one is a 'slave' (receving data from 'master').
Never had deconnection,or unavailability issues with the 'slave'. For the 'master', it is another story... I even replaced it with another D1 and same settings... same unavailability issues ! Both of them are constantly connected to the network anyway. Wifi sleep is deactivated on both of them.
They are both integrated in Home Assistant. The only way/turnaround I found to wake up the master is making a script on a defined effect preset. When the script is fired, the master wakes up and shows immediately.
Hope it helped....

GeckospotNixie, 2021-02-08:
What in the script?
I will give it a go!

spinkohass, 2021-02-08:
Seems like the same issue I have, but I'm using esp8266/nodemcu boards and both are having this issue. I tried disabling sleep mode and dont have alexa sync enabled. They lose connection intermittently. I can see they have IP addresses on the router. What's interesting is that if one drops offline, it can still be controlled via the sync feature on the other controller which suggests that it's the web-server dying? Anyways, this is super frustrating to a noobie like me. I have not tried going to 11.0 yet. I have 2 new esp boards coming today, will see how that goes.

Doyle4, 2021-02-09:
Curious as I have been following this almost everyday, Is there any Devs in here following this? Im unsure who they are by name is all and hoping its not been accidentally missed as from past experience most bugs get fixed pretty quick, unless this is a serious issue? Thanks :)

spinkohass, 2021-02-09:
> Seems like the same issue I have, but I'm using esp8266/nodemcu boards and both are having this issue. I tried disabling sleep mode and dont have alexa sync enabled. They lose connection [...]

I have 4 boards now, all 4 of them have the same issue. I flashed 2 of them with 11.0 and it seems to have fixed one, but that still leaves 3 boards that are dropping offline.

WJCFerguson, 2021-02-09:
Is it fair to say this only affects some people - and above someone is reporting it still happens on one board in 11.0, so it seems very sporadic, like it could be hardware, or some confluence of hardware & software?

spinkohass, 2021-02-09:
> Is it fair to say this only affects some people - and above someone is reporting it still happens on one board in 11.0, so it seems very sporadic, like it could be hardware, or some confluence of hardware & software? 

The 11.0 board just dropped offline. It made it about 24 hours.

SebastianBar, 2021-02-09:
> > Is it fair to say this only affects some people - and above someone is reporting it still happens on one board in 11.0, so it seems very sporadic, like it could be hardware, or some confluence of hardware & software?
> 
> The 11.0 board just dropped offline. It made it about 24 hours.

Go to `0.10.2`, that's the only fix for all of my boards

spinkohass, 2021-02-09:
> Go to `0.10.2`, that's the only "fix" for all of my boards

Thanks! Will try this on the worst offender.

Doyle4, 2021-02-09:
Is it possible to implement the effects over to 0.10.2 and build? (sorry, not good at coding so unsure about this)

spinkohass, 2021-02-09:
> Is it possible to implement the effects over to 0.10.2 and build? (sorry, not good at coding so unsure about this)

10.2 didn't work for me, it's dropped offline twice in about an hour (so worse). I'm trying disabling wifi sleep now to see if that helps.

Urchin73, 2021-02-10:
@GeckospotNixie 

My script for the preset...
Let us know !

alias: preset 8 wled
sequence:
  - service: wled.preset
    data:
      entity_id: light.wled_master_2
      preset: 8
mode: single

WJCFerguson, 2021-02-11:
With a more recent build, I'm now 44 hours and counting with no lockup...

On 0.11.1 I never saw 24 hours - usually I turned it on in the morning, if I was lucky it made it to bedtime, and it was not connectable in the morning.

Then I added a custom IR remote and built from (then) master (94941a7).  I haven't seen a lockup yet - a few ~24 hours when I manually restarted it, and now I'm trying to leave it alone to see whether it remains stable.

Just a single datapoint on one cheap D1 Mini clone.  0.11.1 was my first install, sorry I can't say if earlier releases were OK on my hardware.  I currently have wifi sleep turned off.

Urchin73, 2021-02-11:
What worked for me to wake up an unreachable Wled is accessing it via its ip in the browser. Disregard the 'failed to connect' message and click on the power button left above.

Urchin73, 2021-02-11:
What could be interesting for those of you having recurrent disconnections is to put the device in sync receiver mode (even if there is nothing to receive). I suspect it does something because my 3 receiving devices don't disconnect while sending device does many many times. 
But unfortunately, I can't change it for different reasons on my setup... just trying to isolate the issue step by step.

spinkohass, 2021-02-11:
@urchin73 - What I assume is happening is that the web browser is locking up. The reason I assume this is: 1) I can see that the devices still have an IP address from my routers end. 2) if I have Sync on, the "offline" devices will still receive the sync and update. 

I have not been able to "wake up" the web server by going to the IP address via a browser on a computer - the only way I've been able to fix is to powercycle. Luckly I was smart enough to put a smart outlet above my kitchen cabinets incase this happens.

Urchin73, 2021-02-11:
... strange. Maybe we have different kind of 'unavailability/disconnections' here ?

Travisco29, 2021-02-11:
It's odd sometimes it loses an IP according to my Ubiquiti router. Could
just be Ubiquiti being weird but I don't know.

spinkohass, 2021-02-11:
> ... strange. Maybe we have different kind of 'unavailability/disconnections' here ?

Likely. hah. And just to note - I don't know if these boards have their own issues as I got them and just flashed them to wled out of the box. Also, I'm new to all of this... I have 0 expereience with arduino and these types of things. I have 4 boards (3 deployed to LED strips). I was hoping to flash my last board (Which was the worst offender) to stock arduino and see how it behaves.

Urchin73, 2021-02-11:
Pretty sure the issue doesn't rely on the boards. I considering myself as a newbie as well. Started all this via Home Assistant one year ago.
I did a lot of other projects with these boards and no issue on the +10 I have.
Also Wled deconnection issues are not happening to D1 mini only but also to NodeMCU.
Unsure it is happening to ESP32 tough.
The question about ESP32 was already raised in a previous comment but didn't get any answer. 
It would be nice to have some feedback on ESP32 as well... just to know whether or not we have to consider it as THE board for WLED.

Legsmaniac, 2021-02-11:
> Also Wled deconnection issues are not happening to D1 mini only but also to NodeMCU.

I have to say, I personally cannot agree with this because I've used WLED for over a year now and use both D1 Mini and NodeMCU 8266's and I never have any WiFi disconnection issues.  Sure, there are a small number of people on here complaining of WiFi issues but there are a large number of WLED users.  Yet only a small number of people with WiFi problems.  Which only make me wonder just what is the connection with their own WiFi network?  Sorry, but I can only assume there MUST be something in their/yur setup that causes these disconnections because I and hundreds of other WLED users just don't have any problems so surely there cannot be any issue with WLED and its coding in itself, but with individual users network setups?

Just wish we could get to the bottom of this one.

Urchin73, 2021-02-11:
@Legsmaniac - I follow you ! Most of deconnections issues are related to the network setup in most of the cases for sure.
Using 'deconnection' is maybe not the right word because they are still connected to the network (in my case anyway)... they just become unavailable (some of them). I use 4 D1 with WLED and 3 are running just fine. The 4th one has different settings as explained earlier and is at the same place than the other ones. It has even been replaced with another board... same unavailability issues.
What I was trying to say in my previous comment: unavailabilities could happen with 8266. What about ESP32 ? 
What I **didn't** wrote: running WLED on 8266 will lead to deconnections and unavailability.

WLED is a F.A.N.T.A.S.T.I.C code. The effects are just sooo amazing (fireworks is my preferred one) ! 
Runnig also the sound reactive version of WLED: mindblowing !
Just like you I hope we could get to the bottom of this one ;-)

@Aircoookie - You rock ! Many thanks for your work and your time !

spinkohass, 2021-02-11:
> @Legsmaniac - I follow you ! Most of deconnections issues are related to the network setup in most of the cases for sure.
> Using 'deconnection' is maybe not the right word because they are still [...]

Very well said - agreed on all points.

sansillusion, 2021-02-12:
To add to the topic, I run esp32 on two boards, one controls 61 ws2812b
behind my tv, receiving data from hyperion (lbrelec on rpi4) and the second
drives 21 leds in a custom lamp. Both get polled/controlled by Home
Assistant.

I have had many "connection" issues with both board randomly at different
times. Never having issues with the two at the same time either. I chased
the issue through many versions.

Sometimes little things would help sometimes I couldn't getnitnto be
completely stable.

Now  I have 0.11.1 running on both boards without issues at all "over 3
weeks uptime now"

What I think did the trick and may point to bad/cheap eeprom is, wipe the
eeprom more than once and flash using usb more than once.

I might just be going nuts but I it is now running great and I am almost
afraid to update nex release lol.

sansillusion, 2021-02-12:
Also I have a script in He Assistant that changes color and effect every 45
seconds and it would always accelerate the crashing/disconnecting when I
had issues. Now it's the firs time that I have them running so long and
even use my script a couple times for over 8 hours and no issues. It would
usually bug out in less than an hour using the script before.

>

psinide, 2021-02-12:
Thanks for these tips! I have't gotten around to working on the LED strips again, life getting in the way. I intend to find some time over the upcoming weeks and test out all the solutions mentioned here. My ESPHOME setup is working flawless. I've noticed its a bit laggy with LEDFX, but it could be because I have my instance of LEDFX running on my RasPi 3B as a HASSOS addon.

[36 more replies]

closed 2022-04-27 by stale[bot]