wled/WLED #1630: v0.13.0 send settings data to locIp

WoodyLetsCode, 2021-01-07:
Hey,
I saw that a revamp of settings page is planned for v0.13.0, so here is a suggestion for it.
Currently when you change something in settings the data will be loaded into the same browsing context as the current one. The post request is sent to `http://[current-ip]/settings/leds`.
I would like it, if you handle this like you do in index.htm. If the local storage locIp is set, than use this ip as target. So if locIp is set the post request will be sent to `http://[locIp]/settings/leds`.
This would make it much easier to implement new features and to have better access to settings.

Thank you! 😄

Aircoookie, 2021-01-08:
Hi, that is a great idea, thanks! :)

blazoncek, 2023-08-05:
Done in 0.14-b3.

closed 2023-08-05 by blazoncek