wled/WLED #1632: 3D Printer Motherboard Multi Controller

Doyle4, 2021-01-09:
Hello,

I found my old 3D printer motherboard today MKS GEN L V1.0 8bit board.
Looking over it, it has many 5V and GND's and lots of data pins, Marlin is an open source firmware and unsure if that firmware is capable of controlling LEDS like WLED does?

Lets say it doesn't, is it possible to turn my old board into a giant WLED multi controller? I have a feeling yes but I just have no clue about coding to do so.

Marlin has an LED feature but its super basic, ON/OFF - change colour - brightness but nothing like what WLED can do.

Thanks :)

Doyle4, 2021-01-09:
I also have a spare Wifi Module that can connect direct to the board also.

SpaceCadetLights, 2021-01-13:
Printer control boards have all the outputs you would need, it would just be a case of wiring it to an esp32, and having the appropriate outputs and inputs wired up right. You could easily design and make a pcb to adapt the wiring from an esp32 over to an Arduino mega footprint, or whatever footprint the control board took. 

Very possible, I like where you're head is at :)

Doyle4, 2021-01-13:
> Printer control boards have all the outputs you would need, it would just be a case of wiring it to an esp32, and having the appropriate outputs and inputs wired up right. You could easily design [...]

Thank you! Im really glad someone else understands what I am trying to do here, Over on the Marlin page I'm getting nothing but trouble, saying I'm comparing a Rolls Royce to a Xmas tree.. or they need to contact NASA to borrow a space craft to use as a blender.. so I thank you for taking your time for a positive reply.
Going to look into this more.

Legsmaniac, 2021-01-13:
It certainly would be possible to use that printer board to control LEDs since it is basically an Arduino Mega 2560 on board but not with Marlin firmware.  Whilst I don't entirely agree with the guys over on the Marlin group, I can kinda see where they're coming from.  Why would you want to?  There are better boards out there than the Mega2560 and they are so cheap.  (Way cheaper than printer boards!)  Even boards with WiFi and/or Bluetooth built-in.  If someone really wanted to, it should be possible to adapt WLED to run on the Mega minus the WiFi or even with a WiFi add-on.  But if you're gonna have to buy a WiFi add-on, you might as well buy an ESP8266 or ESP32 instead. The only advantage of the Mega is the number of ports it has for controlling lots of strips but there are boards such as the Wemos D1 ESP32 which isn't far off.
Why not use the board to make another 3D Printer?  Or a CNC Wood Milling Machine?  Or a Laser Engraver/Cutter?  Or a Drawing Machine/Plotter?  😉

Doyle4, 2021-01-13:
@Legsmaniac Thank you also for a positive reply :)

The main reason Id like to use it is as you state, for many of my light projects for personal home use to be controlled via one control unit, I mainly use ESP32's etc for little projects to be sold on. 
As its an old style of board, its now at a stage where latest Marlin firmware struggles to run on it, meaning if I did make another printer or a cnc machine etc, i'm going to be limited and 32bit printer boards are cheap enough to buy than install an old 8bit board.

Marlin would be removed for sure, and I have an ESP wifi module addon also I can use.

Im in no desperate need of it, just a future idea I had :)

Thanks again.