wled/WLED #1635: [DMX] dmxmap not working

MajorTwip, 2021-01-10:
I use an "witty cloud" with default_envs = nodemcuv2. 
Everything works fine! Just not the DMXmap. I do not even know what it's for^^

Below the Pagesource if I open the dmxmap.
The String in the html_other.h in the other hand is complete. 

Do others have the same issue? if yes I'll look deeper into this. If it's only me I won't bother. 

<!DOCTYPE html><html><head><meta content="width=device-width" name="viewport">
<title>DMX Map</title><script>
function B(){window.history.back()}function RS(){window.location="/settings"}function RP(){top.location.href="/"}function FM() {
CN=3;
CS=1;
CG=3;
LC=35;
var CH=[1,2,3,0,0,0,0,0,0,0,0,0,0,0,0,0];
var t=["SET 0","RED","GREEN","BLUE","WHITE","SHUTTER","SET 255","DISABLED"],n=[];for(i=0;i<512;i++)n.push(7);for(i=0;i<LC;i++)for(FS=CS+CG*i,j=0;j<CN;j++)DA=FS+j,n[DA-1]=CH[j];for(DMXMap="",i=0;i<512;i++)isstart="",(i+1)%;margin:0}
</style></head><body onload="FM()"><div id="map">...</div></body></html>

BN3000, 2021-05-27:
Hi, yes I see the same issue. When I press on the DMXmap button I just get a white screen.

BN3000, 2021-09-27:
Little tap to keep the ticket open.

Aircoookie, 2021-09-27:
Thank you for the reminder. Fixed in master now :)

closed 2021-09-27 by Aircoookie