wled/WLED #1638: Error: Which bin for SK6812

freaktech, 2021-01-11:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Aircoookie, 2021-01-12:
The standard one without a suffix should work fine :)

closed 2021-01-12 by Aircoookie

freaktech, 2021-01-12:
tank you

the problem was i used more then 10 segments 
this will happen errors

Aircoookie, 2021-01-12:
Ah, I couldn't know you were talking about that :) I will attempt to fix that one. For now I would recommend to stick to 10 segments or use an ESP32, which _should_ be able to handle 16.