wled/WLED #1639: Segments?

Jabol86, 2021-01-11:
Hello.

I have a WLED with a 570 LED strip. I wanted to find out if it is possible to send a command from HA that will make selected LEDs light up in a selected way?
For example, wants the LEDs from 200 to 350 to light up?

WJCFerguson, 2021-01-30:
I'm not involved with the project - just a new user, but it sounds like you're talking about the segments feature - so it's absolutely possible.

See https://github.com/Aircoookie/WLED/wiki/Segments

Note, I've found some probably buggy, at least unintuitive, behavior with segments if you change it around a lot, but ultimately it works.  (I literally found your question when looking to see what issues exist for segments).

If I'm right and this is the answer to your question, please close this issue to reduce the adminstration effort for the devs.

WJCFerguson, 2021-01-30:
I'm just a new user, not involved in the project, but I'm puzzled by your question - this seems like the basic operation of the device and the Segments feature.

Most likely what you would want to do is define your Segment, use it to define a Preset (which includes Segment configuration) with your desired display, and then use HA to turn on that preset?

Perhaps I'm missing some depth about your question, in which case, apologies, but taking your question "is it possible?" literally, the answer is surely "yes".

closed 2021-06-07 by Aircoookie