wled/WLED #1640: LED string starting 100 LEDs in

abyssdj, 2021-01-11:
This one is puzzling me. I have recently bought a couple of strings of [these](https://www.amazon.co.uk/gp/product/B08G4XRDBR/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) LED "christmas lights". If you've seen them and aren't sure about them, buy them, they're fantastic. Anyway, after taking out an LED mid string, I cut and soldered the string, which works fine on one nodemcu. 

However, on a second nodemcu, the remaining end of the string (minus the bad LED and the odd amount left over), appears to start at LED 101.  It's a string of 100, the second half of a set of 200. If I set the amount of LEDs to 100, it does not work at all. Setting to 200 and segmenting 100-200, it works perfectly. I can't think of a reason this would happen! If I swap and connect a strip of 10 standard ws2812b, it works as expected. What gives? I may be missing something simple, but so far this is puzzling the hell out of me! Surely these LEDs don't have any kind of memory or fixed address???

abyssdj, 2021-01-11:
Sure enough, the odd offcut seems to respond from 88-100! What on earth is going on here?

sansillusion, 2021-01-11:
These are hard adressed, the led has an adress coded innthe chip at
manufacture.  There was a post/issue about that somewhere here. I dont
remember if there is a way to use them other than setting leds count to 100
and setting the segment to start at 87 and end at 100.

abyssdj, 2021-01-12:
On the one hand, I'd like to argue that as I can't see the logic in them hard coding an address, plus, to my knowledge (and do correct me if I'm wrong) but whilst being classed as addressable, ws2812b (etc) don't actually have an "address" as such at all, do they? They're basically just shifted out as far as I know?  On the other hand, they do seem to be acting exactly as though they were hard coded. I just don't see how an LED without an address (ie standard ws2812b strip) and an LED supposedly with an address (ie, this string) would function so differently when run on the exact same protocol, same nodemcu etc.

Either way, if that's how they are destined to function, segmenting is just a slight inconvenience, not a big deal at all considering how well they actually work. That is until we start looking at sacn/artnet/etc in which case WLED ignores segments. That could become an issue as in this case it would immediately block 300 channels (the first 100 "missing" LEDs). Maybe rehash the "skip first led" option into "skip first x amount of leds"? so that issues like this can be worked around, @Aircoookie ?

Aircoookie, 2021-01-13:
Those for sure are very unique and don't work the same as regular WS2812b - which are just a glorified shift register as you said. More details at https://github.com/Aircoookie/WLED/issues/1312 👍 

The hardcoded addressing mostly seems like an inconvenience, it would seem to make it essentially impossible to daisy chain them or to remove and reuse a section like you did. However, it can also make some unique use cases possible like running LEDs 0-50 and 50-100 from the same LED pin into two different directions and still having them address correctly.

I might consider allowing more LEDs to be skipped in the future. Segments are a good workaround especially considering presets can make use of them now, but you are correct about the E1.31/artnet problem

Legsmaniac, 2021-01-13:
Yeah.  To be fair, those LEDs aren't designed for general hobbyist use after all anyway but created created for use with specific products with their own controllers.  Though why they need to be hardcoded, I cannot imagine.  I'm guessing there must be a reason?  A shame really because they can be bought so cheaply.  Or maybe that's the reason, to prevent people buying for use with other projects.  Wonder how long it will be before someone comes up with a program to recode/hack them?  IF it's possible of course.

abyssdj, 2021-01-13:
Thanks for the info. After reading through the other thread, it does all definitely seem to add up! I wouldn't say it's all bad though, I think that kind of topography could be very useful in some cases. It also does stop a dead LED taking out the string - I know that from experience! Definitely need that LED skip setting though for E1.31/Artnet. Could that be coded into a usermod?

Why they're addressed in the first place is anyones guess. How is more interesting though.... I'm sure they're not done at the time of manufacture, that would just seem senseless to me as they'd have to have 200 "different" LED's to pick from. I expect, as others have mentioned, that there's a way of writing to them somehow, as they offer the strings in different lengths. I expect they build the string, blast it with LED addressing data, and then attach the USB controller. Or, maybe the controller is the key?

abyssdj, 2021-01-13:
By the way, does anyone know the power consumption of these LEDs?

abyssdj, 2021-01-19:
Small note on this issue - the issue can be "solved" when using e1.31/artnet/etc by using the realtime led offset feature. Offset by the amount of "missing" LEDs and hey presto, your string of weird LEDs will start at channel 1.

closed 2021-06-09 by stale[bot]