wled/WLED #1641: [UDP] switch LEDs synced on/off [ambilight]

schwatter, 2021-01-11:
So first, thank you for this great piece of software.

Since StefanRu1 write some pythonscript for wled to pipe data , we all can use enigmalight to use wled
over the air.

https://github.com/StefanRu1/WLED-Enigmalight/tree/main/Enigmalight%20Plugin

So i modified enigmalight to handle wled to set different states. It works perfect with one instance, but enigmalight 
can handle more. 

So the goal is always to set on/off synced devices, no matter if udp is on or off.

I first tried to handle all with some delay around 10 sec. So enigmalight goes in sleep and udp is cut. Then wled sync
is up and i can handle synced devices. But i get in trouble when i start from sleep. My delay has no effect.
So yeah, enigmalight is also a bit moody... Go to sleep by the way is working this way. 

So today i thought about reading in the enigmalight.conf with all wled_ip's. Looks first right. Using all ip's and send
httprequest to set on/off all wled_ip's after another. But i think, the next trouble is on the way. I would say shut off
will work, because the devices are not synced. But when they are synced this could be discolights?  :-)

The guys from hyperion by the way have the same problem.

Aircoookie, 2021-01-13:
Hi!
Maybe it would make sense to allow UDP sync to work in realtime mode - I have to look into it.
Sending the HTTP with an OFF command to all units sounds good - if the sync started working before the HTTP command was received, the light would just turn off by whatever is received first - no discolights and no problem :)

StefanRu1, 2021-01-13:
Hi Schwatter,
cool that you enhance the features for EnigmaLight.

Just a remark:
You have seen that i have the timeout set to 5 sec at the moment?
colorbyte = bytearray([2,5]) # DRGB Header, 2 for DRGB Protocol, 5 for 5 Sec until return to normal mode

So if realtime UDP does not send data it will take 5 sec until it returns to normal mode.
Should i reduce this?

Best regards,
Stefan

schwatter, 2021-01-13:
@Aircoookie

The problem triggered me yesterday, that i read now the config in and filter for all ip's. After i get the pool, i fire
the httprequests (on/off/dim/toggle) to all wled devices. No matter if they are synced or on udp. Performs good.
But yeah, right out of the box, i mean direct wledsupport, would be the best. One master, one ip, one to rule them all.

@StefanRu1 
Thx for the hint. Will tinker a bit more with the scipt it self.

schwatter, 2021-01-14:
@Aircoookie 

Ok, now i have a good argument, that you build that feature in.

The last days i tested all time with two wled instances and enigmalight. Today i just switched the conf to my main device only 
behind my tv. So my wledautomatic read the conf in and i get just one ip. But after the led's switched on, the second light
also started. 
So the think is, i have here two devices, both are synced and when one is captured from udp the second follows all settings
like brightness, on, off except udp.
On the other site, if i change something in the second device nothing change in the udp captured device.

This happens, when both devices are synced on the same port.

blazoncek, 2023-01-25:
UDP and realtime syncing is possible since 0.13

closed 2023-01-25 by blazoncek