wled/WLED #1642: WLED does not turn on normally

blogaster, 2021-01-12:
Hello! I have an ESP12 firmware WLED_0.11.1_ESP8266.bin.
If the LED strip is connected to the microtroller and I apply power, then one or two LEDs light up and wifi does not work.
If you first apply power and then connect the LED strip, then everything works fine.

LED strip is powered by 5 volts. Power supply unit for 12 volts 2 amperes. Step-down board from 12 to 5. For esp I use a step-down regulator up to 3.3 volts

**WLED version**
 - Board: ESP12
 - Version 0.11.1
 - Format bin

Videos and photo: https://yadi.sk/d/LgUEsi4P95qbxw

Thank you for your help!

sansillusion, 2021-01-12:
Soumds like mabe power supply/stepdown too weak for powering the esp wifi
and lights at the same time (probably just peak current).

blogaster, 2021-01-12:
> Soumds like mabe power supply/stepdown too weak for powering the esp wifi and lights at the same time (probably just peak current). Le mar. 12 janv. 2021 4 h 05 a.m., Ilya Blogaster <notifications@github.com> a écrit :
> […](#)

It's strange. I am using 12V 2A power supplies. Tried 3a and same problem.
I connected a multimert and noticed that if you sharply increase the brightness of the white light, then at 8 watts everything reboots.

Mini560 step-down board. Its working current is 4 amera and 5 in the peak

sansillusion, 2021-01-12:
The stepdown might not be stable enough for fast changes, try adding large
capacitor next to the esp mabe if you can.

blogaster, 2021-01-12:
@Aircoookie helped me!
 Sounds like you might need a 330 or 470 ohm resisitor between your data and 3.3v pin
It's work!

closed 2021-01-12 by blogaster