wled/WLED #1645: Sonoff Wled

arnaud-aub, 2021-01-12:
### Bonjour à tous,
J'ai plusieurs relais sonoff et j'aimerais pouvoir mettre le firmware Wled dedans, ce que j'ai fais mais je n'arrive pas à le trouver sur _mes réseaux disponibles_. Pouvez-vous m'aider ? Est-il possible d'utiliser un sonoff avec WLED ? Je vous remercie de votre compréhension. 
Ps : **j'utilise un sonoff basic R2 V1.0**

Aircoookie, 2021-01-12:
Bonjour!
Yes, WLED should work fine on a Sonoff device if you have used the `WLED_0.11.1_ESP8266_1M.bin` binary file.
Have you already connected your LEDs, and if yes, do they light up?

arnaud-aub, 2021-01-13:
J'ai connecté le data de mes leds au GPIO14 et le GND à son alimentation ainsi qu'au sonoff et le VCC à son alimentation, mais aucune led ne s'allument, est-ce normal ?

Aircoookie, 2021-01-13:
Hi, yes this is normal because the default LED pin is GPIO2. I've compiled a custom binary for pin GPIO14, if you flash this the LEDs should work correctly :)
https://drive.google.com/file/d/1lapHm-KEKJ-z7FkuEOg5Ax9FlxUgVyOe/view?usp=sharing

arnaud-aub, 2021-01-13:
Salut bonne nouvelle tout fonctionne comme je le voulais, merci. Mais j'ai une question, étant donné que c'est un sonoff j'aimerais savoir si il est possible de commander le relais sur le GPIO12 ? Merci

sansillusion, 2021-01-16:
Il y as un usermod qui permet de contrôler un relais.

https://github.com/Aircoookie/WLED/tree/master/usermods/blynk_relay_control

Tu dois compiler toi même wled pour cela.

arnaud-aub, 2021-04-15:
bonjour, cela fonctionne avec des pixels WS2811 mais pas avec un ruban led WS2811 pourquoi ?

closed 2021-08-28 by stale[bot]

arnaud-aub, 2021-11-23:
Bonjour, je reviens vers vous après un grand nombre de test et je ne comprends pas, WLED sur SONOFF fonctionne avec des pixels WS2811 mais pas avec un ruban led WS2811, sur le ruban, les led ne fonctionnent pas correctement pourquoi ? Merci

PS: j'ai déjà modifié tous les paramètres et rien ne change.