wled/WLED #1646: Hide AP not working - again?

Legsmaniac, 2021-01-12:
**Describe the bug**
Despite ticking Hide AP name in WiFi setup, the AP name is still visible.

**To Reproduce**
Tick the box Hide AP name and check your list of available WiFi Networks.

**Expected behavior**
AP name should remain hidden.

**WLED version**
 - Board: Wemos D1 8266
 - Version  0.11.1, dev2101040
 - Format  self-compiled binary via Platformio

**Additional context**
I reported this issue before Christmas and it was fixed. 
Something has obviously since changed as the bug is back?

Thank you for your help!

Aircoookie, 2021-01-12:
Hi!
I looked through the code quite a bit thinking we'd have a regression on our hands.
Seems like the issue is likely a different one - if the AP is already running, it will not re-initialize after saving WiFi settings. So most AP-related settings (SSID, pass, channel, hide) will only take effect after a reboot. I'm not sure whether I should classify this behavior as a bug, because re-initing the connection every time would likely disconnect your client device. On the other hand, you would expect the settings to apply immediately (which some WLED settings do, some don't - I'm planning to add  _Reboot required to apply changes._ labels in a future update for those settings.
Does the AP stay hidden after a reboot?

Legsmaniac, 2021-01-13:
Pretty sure I had rebooted several times but just in case, I rebooted again.

And apologies - it's gone now. Hidden. 😬

Lockdown must be getting to me........

closed 2021-06-09 by stale[bot]