wled/WLED #1647: How to flash ESP32 with W25Q128 (16Mb) ?

athlon1900, 2021-01-13:
Hello !

I like to flash a ESP32 with W25q128 , but it fail.
Bootloader at 0x00000 and Image at 0x10000
On a W25q32 flash-chip the bootloader and image working great with these settings.

Do i need another adress for W25q128 ?
If yes , which adress for bootloader and image should be set ?

Thanks , athlon1900

Aircoookie, 2021-01-13:
Hi!
What fails, the flashing progress, the lights and/or the WLED-AP showing up? What is your flasher program?
The partition layout is only setup to use 4mb, but it _should_ run fine on 16mb units. Have you tried erasing the flash?
Hope we can get it to work! :)

athlon1900, 2021-01-14:
> Hope we can get it to work! :)

Hi Aircoookie !
Thanks for your answer. I hope so too :)

Flashing the ESP32 is done by **flash_download_tool_v3.8.5**
Settings are : SPI SPEED = 40Mhz ; SPI MODE = DIO ; Flash Size = 128Mbit ; 
                      esp32_bootloader.bin at 0x00000 and WLED_0.11.1_ESP32.bin at 0x10000
I always click **ERASE** befor flashing. ;)

The flash-procedure is done with no errors , but after restarting the device the bootlog says : Waiting for download
It seems that the device didn't find a firmare on the flashchip and boots into flash-mode.

The second device i have , has a 32Mbit flash , with the same settings but FLASH-SIZE = 32Mbit , the ESP32 boots the WLED-Firmware.

What is going wrong ? What can i do ?

> Hope we can get it to work! :)

athlon1900, 2021-01-15:
Hi !
I have got it working .
Flashing was easy , if you know it :) ( Google was my friend , but hard for me to find )

My steps : 
1. Erasing the flashchip
2. In the **flash_download_tool_v3.8.5** there is a folder named "** bin **". In this folder there are four binaries named **_a_0x1000.bin** , **_b_0x8000.bin** , **_c_0x10000.bin** and **_d_0x225000.bin** . I think the four binaries a,b,c,d are for initial-flashing.
3. Setting the flash address of the files : for **_a_0x1000.bin** set to **0x1000** , and so on. 
4. Then i flashed the four files.
5. After flashing i set the esp32_bootloader.bin to 0x00000 and the WLED-firmware WLED_0.11.1_ESP32.bin to 0x10000
6. Flashed these two files , did a reset on the ESP32 board and the firmware was booting . :)

Now the ESP32 Board with the W25Q128 (16Mb) is working without any Problems.

Thanks for help ,
athlon1900

Skywalkerf34, 2021-03-04:
Hi  athlon1900,

I am acing the same problem, And I have this download tool same version, but unfortunately nothing in the BIN folder.
So I check and Download again from espressive but their ZIP didn't contain anything in BIN folder !
Where did you get your tools or those bin ?

Rgds

athlon1900, 2021-03-05:
Hi Skywalkerf34 !
I am amazed that the files are no longer included.
Here are the files.
[bin.zip](https://github.com/Aircoookie/WLED/files/6091530/bin.zip)

Best regards

Skywalkerf34, 2021-03-05:
Hi athlon1900,

thanks a lot, I will check next step tmr

Best rgds

Skywalkerf34, 2021-03-06:
Hi athlon1900,

GOOD NEWS / IT MAKE IT WORKS !!!!!! 
thanks a lot,

Best rgds

closed 2021-03-06 by Aircoookie