wled/WLED #1652: Feature request: AnalogClock custom colors (and brightness) for hrs/min/sec and 5min dots

DDNRN, 2021-01-13:
I use analogclock overlay at some devices and have a design problem with fixed colors for the clock dots.
Using another affect in the background is possible but the colors don't meet the clock and even not to the "housing" this clock is set in.
Must be not to hard to implement a settings value for this 4 color codes?

Another wish on top is to combine the colors if one dot overlays another from the clock dots. Always searching for the mins or hours if one erases the other one at same minute LED. Easyest way is to add color codes together to have mixed colors.   

. . and thanks to all that work about WLED, its a fantastic project!

blazoncek, 2023-01-25:
Done in recent updates and new usermods as well.

closed 2023-01-25 by blazoncek

jasmohan-narula, 2024-04-17:
I also need this feature request. Can anyone explain the how to set custom colors for clock? I'm unable to find the settings.