wled/WLED #1660: ESP HW remote

b3-4r, 2021-01-16:
Hi, we've been using WLED-controlled lighting features throughout our household and they're perfect for setting just the right mood, be it standalone or synced to music over E1.31. It's, however, proven annoying to have to use a phone or computer to change the FX. I've been wondering if it would be possible to create a simple, standalone, ESP8266 or ESP32 based remote with a set of buttons, or perhaps even an encoder and a small I2C OLED display, to control the brightness, select FX and colour palette or even set the timer on individual WLED controllers in the network, or all together. The IR remote option is cool but doesn't work reliably when the IR RX is hidden behind furniture etc.

Another idea would be to use the [M5Stack with the encoder Faces module](https://m5stack.com/products/encoder-module), featuring a large knob with an RGB ring indicator for instant feedback (hue, sat, int values could be visualised, for example).

Thank you for considering this and for the spectacular work on WLED!

smartroad, 2021-01-16:
I would love this as well. I started designing a circuit for running my lights (before I found WLED) using a custom Arduino which was going to use exactly your spec, encoder and OLED. I'm using a NanoLeaf 3D printable clone and was going to make a blanking plate for it to mount to so you can turn on/off the display and select different modes - means anyone can adjust it if needed.

sonnyff, 2021-01-21:
A project I had in mind was repurposing a midi controller full of knobs and sliders to control my smart home. My plan was to connect the controller to a raspberrypi, and use midi2mqtt software to pipe all the outputs to Node Red, from there it would be trivial to control WLED, though this lacks feedback.

SpaceCadetLights, 2021-01-21:
I am working right now on designing and building a capacitive touch remote control that is designed for wled. I have made a simple control board pcb that you plug your esp32 into and it has a power input, 8 led strip outputs, mic input, ir input, and button input. It also has an nRF24L01 radio module on board and communicates to the remote over radio. I am working on the usermod right now for it, and waiting on the circuit boards to come in for the remotes themselves. Once finished you can sync multiple remotes to the esp by pressing a button in the app, and then a button on the remote. The remote can control speed, intensity, brightness, power on and off, select between 1-15 presets, and toggle auto cycle mode. You can also modify the code to make the remote control any other part of wled. I'm nearly finished with it and I'll post the code and pictures once complete. I'm also planning on selling the control board and remote as a set If anyone wants any :) also the remote has a magnetic wall mount that screws to your wall for easy storage and is powered by a couple AAA batteries. 

I'm designing this to be very modifiable so you could make the remote control whatever you wanted in wled by modifying a few lines of code. I will also be making wall mount capacitive touch control panels that resemble a light switch panel so that you can integrate them more easily into your home. 

The rf modules have great range, so hopefully this will help solve this issue :)

b3-4r, 2021-01-23:
@sonnyff that's the beauty of open standards for control but also its downfall, as you need a RasPi running somewhere, perhaps the controller wired into it- I can see this being used in a club setting

@SpaceCadetLights I shall follow your repos

The more I've been thinking though, the idea of M5Stack w/ encoder module seems so appealing, too bad I can't program in the slightest. I think I'd even pay for something like that, list of lights, tick or untick ones to be set, either choose a preset/ intensity/ dimming or go through the options, hue selection with a rainbow ring around the encoder, white or off where selected; white->sat colour ring for sat, off->full ring for intensity, then a list of effects, plus the standard params; clock around the encoder for timer settings etc. Plus the package is super sleek.

sonnyff, 2021-01-23:
> @sonnyff that's the beauty of open standards for control but also its downfall, as you need a RasPi running somewhere, perhaps the controller wired into it- I can see this being used in a club setting

Use a raspberry pi z with wifi, grab a battery bank, and stick them to the back of the controller. :)  
The controller I have is keyboard shaped, so a pi and battery won't add too much more to it.

sonnyff, 2021-01-23:
@b3-4r Use a raspberry pi z with wifi, grab a battery bank, and stick them to the back of the controller, totally portable :)
The controller I have is keyboard shaped, so a pi and battery won't add too much more to it.

dutchmendesign, 2023-01-05:
@b3-4r I have the same idea! Was thinking of making a remote usermod_v2_rotary_encoder_ui using ESP_NOW
https://github.com/Aircoookie/WLED/tree/main/usermods/usermod_v2_rotary_encoder_ui
https://www.youtube.com/watch?v=8IGkFBghzww

blazoncek, 2023-09-19:
It is possible (since 0.14.0-b4) to use WiZ Mote wireless remote with WLED.
Works with ESP-NOW technology.

There is also IR remote support built in and usermods for rotary encoder (with or without OLED displays).

closed 2023-09-19 by blazoncek