wled/WLED #1664: Send JSON Packet via MQTT

Deggle, 2021-01-16:
The MQQT interface allows the HTTP style packets to be sent, however not JSON. Could an MQTT topic such as /json be added to simply receive packets and send them off to the existing desterilise function?

My use case is that I've found MQTT often seems to be a little more stable and less laggy than making HTTP POST requests (generally, I find this on a number of ESP based things I use). Another use case would be the ability to send JSON blocks (e.g. setting segment colours) to a number of different devices using the group topic.

Legsmaniac, 2021-01-16:
> My use case is that I've found MQTT often seems to be a little more stable and less laggy than making HTTP POST requests

Strangely, in my case, I found exactly the opposite.   I guess it could depend on what we're sending the requests from and how they are coded.  I have two projects where I need an ESP8266 to send HTTP requests to another ESP running WLED.  For me, MQTT was a bit hit and miss.

Aircoookie, 2021-01-17:
Hi! Since 0.11.0, this is in fact supported, the `/api` topic accepts JSON API commands. The only reason why MQTT doesn't officially support JSON is that it doesn't publish the state yet. I updated the wiki to reflect this :)

Deggle, 2021-01-17:
That's great - tested and working!

closed 2021-01-17 by Deggle