wled/WLED #1665: Leds wont turn on even though Nodemcu is configured correctly and interface connects to nodemcu

Aviv-Hoitash, 2021-01-16:
I configured my lights like instructed and everything worked fine, until one day the connections disconnected and when I tried connecting them back the lights wouldn't turn on. The little blue light on the nodemcu however, turned on, and I could connect to it using the interface, but when I changed the colors and tried turning the led's on, nothing happened. This is weird as my lights have worked before so I don't know what I'm doing wrong.
**WLED version**
 - Board ESP32
 - Version 0.11.0,

Thank you in advance for your help!

DivanX10, 2021-01-27:
despite what the address of the ribbon you are using. If ws2812 and if the first diode of this tape burns out, then the other diodes may not work. For example, you turn on the diodes through the web interface and the web interface itself shows that the diodes are turned on, but the diodes themselves do not work. This means that most likely the first diode failed. Check with other diodes.

Aviv-Hoitash, 2021-01-27:
> despite what the address of the ribbon you are using. If ws2812 and if the first diode of this tape burns out, then the other diodes may not work. For example, you turn on the diodes through the web interface and the web interface itself shows that the diodes are turned on, but the diodes themselves do not work. This means that most likely the first diode failed. Check with other diodes.
> 
> смотря какую адресную ленту вы используете. Если ws2812 и если у этой ленты сгорает первый диод, то остальные диоды могут не работать. К примеру вы через веб интерфейс включаете диоды и сам веб интерфейс показывает, что диоды включены, но сами диоды не работают. Это значит, что скорее всего вышел из строя первый диод. Проверьте с другими диодами.

How do I check with the other diodes? I can't cut the first one it has the connections.

DivanX10, 2021-01-27:
do you have a spare address tape, such as the remainder of the main feed? You can certainly step over a couple of diodes and solder them to the tracks, but I would still check on the residual tape. Residual tape, this is the tape that turned out to be superfluous, and you use the desired length for its intended purpose.

есть ли у вас запасная адресная лента, например остаток от основной ленты? Можно конечно перешагнуть пару диодов и припаять к дорожкам, но я бы все же проверил на остаточной ленте. Остаточная лента, это та лента, которая оказалась лишней, а нужную длину вы используете по назначению.

closed 2021-06-09 by stale[bot]