wled/WLED #1666: Watchdog trigger on mode chase_flash_random

nflug, 2021-01-16:
**Describe the bug**
if I let run the mode: chase_flash_random long enough (like 5mins) I get a watchdog trigger error

**To Reproduce**
Start the chase_flash_random mode and wait

**Expected behavior**
Should not trigger the watchdog

**WLED version**
 - ESP32 dev
 - Newest stable

**Additional context**
The error says that it triggers here:

```
uint16_t WS2812FX::mode_chase_flash_random(void) {
  uint8_t flash_step = SEGENV.call % ((FLASH_COUNT * 2) + 1);

  for(uint16_t i = 0; i < SEGENV.step; i++) {
    setPixelColor(i, color_wheel(SEGENV.aux0)); <--------------------------- HERE
  }

  uint16_t delay = 1 + ((10 * (uint16_t)(255 - SEGMENT.speed)) / SEGLEN);
  if (flash_step < (FLASH_COUNT * 2)) {
    uint16_t n = SEGENV.step;
    uint16_t m = (SEGENV.step + 1) % SEGLEN;
    if (flash_step % 2 == 0) {
      setPixelColor( n, SEGCOLOR(0));
      setPixelColor( m, SEGCOLOR(0));
      delay = 20;
    } else {
      setPixelColor( n, color_wheel(SEGENV.aux0));
      setPixelColor( m, SEGCOLOR(1));
      delay = 30;
    }
  } else {
    SEGENV.step = (SEGENV.step + 1) % SEGLEN;

    if (SEGENV.step == 0) {
      SEGENV.aux0 = get_random_wheel_index(SEGENV.aux0);
    }
  }
  return delay;
}
```

Right now I could temporary fix the trigger with this inside the service loop
```
TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
TIMERG0.wdt_feed = 1;
TIMERG0.wdt_wprotect = 0;

TIMERG1.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
TIMERG1.wdt_feed = 1;
TIMERG1.wdt_wprotect = 0;
```
and this in the fx.h
```
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"
```

Thank you for your help!

Aircoookie, 2021-01-17:
Thank you very much!
I suspect this might be due to the `delay` between the calls to the FX functions being to small, where this could happen if the speed is high and/or there are a lot of LEDs. How many LEDs do you use and do you remember the effect speed?
I'm attempting to run this effect on max speed now to see if I can reproduct the wdt reset.

In any case I should rework Chase Flash Random to use the fixed `FRAMETIME` refresh rate :)

nflug, 2021-01-18:
I use 300 leds of the type SK9822 with half of full effect speed. It seems that it happens also with other effects. I wrote a loop that changes the effect every 10 seconds. If the loop runs for over 20mins even the solution I gave above doesn't help only to disable the watchdog completely helps in this situation. 

What I also tried is to add the effect loop to another freeRTOS task and give it vTaskDelay(1), yield(), enough stack size & priority but nothing helps.

closed 2021-06-09 by stale[bot]