wled/WLED #1670: After reboot 60 LED ws2812B becomes 50

ghpk, 2021-01-17:
Hi,
I made a corner Lamp using 1 meter WS2812B LED strip which has 60 leds, its running on NodeMCU.
Wi-Fi is fine, power I am using is 5 amp smps based and very stable on other projects too.

Whenever I turn wall outlet off and turn it back On only first 50 bulbs start working showing the effect I set as default.

To resolve it, I go to LED preference and adjust LED Count
either to 59 or 61 and it starts working again.

Am I missing something ?

Haven't used bug option as there are so many experts here to share much more valuable information and they need to cross verify and may already have a solution.

WoodyLetsCode, 2021-01-19:
Do you use segments?

ghpk, 2021-01-23:
No segment is Not used.

cmkiefer, 2021-01-31:
Any resolution to this? I am seeing the same thing. On power up, not all pixels are illuminated. I haven't counted to see how many or if it's consistent. Edit LED Prefs and change quantity, save and change back to the correct number of lights and all is good...until next power up.

ghpk, 2021-01-31:
Cant still find the solution.
I've made 2 sets of 60 Ws2812b Leds, Each Running of its own NodeMCU board and its own 5 AMP power supply same model.
Boards are same maybe cheap clones but only one has issues.
Every Reboot affects only one lamp.
Have reflashed but to No avail.

cmkiefer, 2021-01-31:
I was suspecting the board as well. Cheap D1 mini I used to replace another that had failed. I think I will replace that and see what happens.

ghpk, 2021-02-04:
I think I resolved this un-confirmed issue.
This is mostly happening when you configure WLED initially as a smaller number of LEDs.
example: make a lamp of 30 WS2812b then you set it up to serve 60 LEDs, this new setting even after save somehow gets jumbled up internally, LED count will continue to show your set value but in bag-round this value gets altered on reboot.

To Resolve, Go to Config under security and updates select "Factory reset" then re-configure with correct number of LED as you do on a freshly flashed Board, Do not restore any available backup if prompted.

Have tested on NodeMCU board and works fine.

cmkiefer, 2021-02-06:
Here's what I've found. It was in the saved #1 preset which I load at startup ("Apply preset x at boot" in LED Setup). Maybe this is new but it evidently saves the LED count from LED Setup into the preset when you create it. Once I fixed the LED count AND Recreated the preset it boots with the correct number of lights... so far.

coryking, 2021-05-24:
I'm running 0.12.1-b1. I've erased the flash, started from scratch but no matter what I do I keep getting 30 count on my first LED strip. This is on an esp32 board. It remembers the pin, but not the count.

coryking, 2021-05-24:
This has to be some kind of bug / regression. I cannot repro this on 0.11.0.... the count sticks.

For what it's worth, I even got adventerous and changed the "default" leds in wled.cpp and wled_eeprom.cpp and it still kept going back to 30 after a hard reboot.

closed 2022-04-16 by stale[bot]