wled/WLED #1671: Wifi off/dropped when WLED 'Power' is on

pwalther, 2021-01-17:
I have a NodeMCU running WLED 0.11.1

When I use the web interface to turn turn the WLED "power" on, it will turn on my LEDs but it will disconnect from wifi. When I push the "FLASH" button on the board, the onboard LED will turn off the LED and wifi will works again.
I am powering it via VIN with 5.25V using a 30A (or so) PSU.
This issue persists even without the DATA pin connected (and restarted).

When I use a PC to power the nodeMCU (power via USB, no LED pin connected), it is reachable via wifi just fine - "power" on or off.

Should I just add a voltage regulator and use a USB plug?

Faszakasza, 2021-02-03:
Did you find a solution to this? I'm seeing the same problem, however I have 4 nodeMCUs and only 1 of them is doing this, the other 3 works just fine. I have already tried all the wifi related tips and tricks, none of them made a difference.

pwalther, 2021-02-13:
OK so in my case this was caused by weak wifi. I added another AP nearby and the problem is gone. What was interesting is that it only happened when I turned the LEDs on, so maybe the power supply introduced some EMI?

closed 2021-02-13 by pwalther

Legsmaniac, 2021-02-13:
> What was interesting is that it only happened when I turned the LEDs on, so maybe the power supply introduced some EMI?

Or maybe it caused a drop in voltage provided to the ESP?  Are you able to measure the voltages if you have a meter?

pwalther, 2021-02-13:
Hmm it could have dropped momentarily, I did measure but just with a regular multimeter, no scope or similar. And with the LEDs on, I measured 5.00v - I even adjusted it to 5.8v. The exact same setup now works after having the AP nearby.