wled/WLED #1673: LED Effect Maker

Doyle4, 2021-01-18:
Id love to make my own effects and add them to my WLED.

Is there any software that lets you create effects and output as code?

Thanks :)

fishbone-git, 2021-01-18:
Hi,

I've seen software to create specific effects to run with other software, but I am not aware of any that generate code.  Probably woudn't be as efficient as you might want to use in the esp8266.  I've mocked up a couple of effects myself as PRs.  I'm always interested in new and interesting effects.  Maybe if you make an animation of the kind of thing you want, it may be simple enough that I can help, or someone else with greater coding skill will fell motivated to dive in.

Doyle4, 2021-01-18:
Thanks for replying Fishbone :)

Cool, I'll stop trying to find some software then haha.
Theres a couple ideas I have, but I need a layout of my LEDS first before I can properly understand what It is I'll need, but its good to know I can ask you :) Thanks.

WolfsWelpe, 2021-01-18:
Hello :)

Actually there is one but sadly not for WLED.
I found this LAMBDA LED Strip Effects generator: https://adrianotiger.github.io/Neopixel-Effect-Generator/
You could probably edit the Arduino Code of WLED and add your own effect into it.

Doyle4, 2021-01-18:
@WolfsWelpe 
Thank you! will take a look thanks :)

fishbone-git, 2021-01-19:
Issue #1678 appears to mention a system which does support creation of effects.  It may or may not be in a form that is most useful for you.

Doyle4, 2021-01-19:
@fishbone-git Thank you, will take a look thanks!

closed 2021-06-09 by stale[bot]