wled/WLED #1674: Горят первые диоды при выключенном состоянии. \ The first diodes are lit when switched off

DivanX10, 2021-01-18:
**WLED version**
 - Board: [ESP8266 NodeMCU_V3]
 - Version [0.11.1 build: 2012180]
 - Format [WLED_0.11.1_ESP8266.bin]

В выключенном режиме через какое-то время включаются первые диоды. На ESP32 такого не наблюдается. Что это может быть и как исправить? У меня на D4 подключен резистор на 470 OM. Так пишут [здесь](https://alexgyver.ru/ws2812_guide/)

"Если в схеме возможна ситуация, при которой на ленту не будет подаваться питание 5V, но будет отправляться сигнал с микроконтроллера – лента начнёт питаться от дата-пина. В этом случае может сгореть как первый светодиод в ленте, так и пин контроллера. Не испытывайте удачу, поставьте резистор с сопротивлением 200-500 Ом. Точность резистора? Любая. Мощность резистора? Любая. Да, даже 1/4."

-------------------

In the off mode, the first diodes turn on after a while. On ESP32, this is not observed. What can it be and how to fix it? I have a 470-OHM resistor connected to D4. So write [here](https://alexgyver.ru/ws2812_guide/)

"If there is a situation in the circuit in which the tape will not be supplied with 5V power, but a signal will be sent from the microcontroller – the tape will start to be powered from the data pin. In this case, both the first LED in the ribbon and the controller PIN may burn out. Do not try your luck, put a resistor with a resistance of 200-500 ohms. The accuracy of the resistor? Any one. Resistor power? Any one. Yes, even 1/4."

[image]

Aircoookie, 2021-01-19:
Hi! This might be a problem with the LEDs picking up noise and interpreting it as a valid data signal, since the ESP8266 doesn't send any data when the LEDs are off. How long is your data wire? You might need to add a levelshifter like the SN74HCT125N. You could also use a relay to shut off the 5v power to the LEDs. When connected to pin GPIO12, this is natively supported in WLED.

DivanX10, 2021-01-20:
Подключил ESP32. Прошли сутки и такого не наблюдаю как, если подключил бы NodeMCU V3. Первые диоды не горят если плата ESP32 находится в ждущем режиме. Блок питания у меня 12а - 60вт - 5в. Буду наблюдать дальше, нужно больше времени. 

Connected ESP32. A day has passed and I do not observe this as if I would connect NodeMCU V3. The first diodes do not light up if the ESP 32 board is in standby mode. The power supply I have is 12A-60w - 5V. I'll keep watching, I need more time.

**Подключено у меня вот так**
**I have it connected like this**
[image]
[image]
[image]

DivanX10, 2021-01-26:
Прошло несколько дней как использовал ESP32, на 3-ий день загорелись первые диоды. В итоге я переделал проводку согласно этой [статье](https://alexgyver.ru/ws2812_guide/).  Решил использовать NodeMCU v3. Пока первые диоды не загораются 2-ой день. Ранее при использовании NodeMCU v3 первые диоды загорались через несколько минут, чего я не наблюдаю. Возможно мне удалось решить проблему, но пока продолжаю наблюдение. 

It took a few days as I used ESP 32, on the 3rd day the first diodes lit up. As a result, I redid the wiring according to this [article](https://alexgyver.ru/ws2812_guide/).  I decided to use NodeMCU v3. Until the first diodes light up on the 2nd day. Previously, when using NodeMCU v3, the first diodes lit up after a few minutes, which I do not observe. Perhaps I managed to solve the problem, but for now I continue to observe.

Дополнительно к установленному резистору в 470 ОМ, я скрутил проводку таким образом

In addition to the installed resistor of 470 ohms, I twisted the wiring in this way
[image]

2.1) Если между лентой и контроллером (Arduino) большое расстояние, т.е. длинные провода (длиннее 50 см), то сигнальный провод и землю нужно скрутить в косичку для защиты от наводок, так как протокол связи у ленты достаточно скоростной (800 кГц), на него сильно влияют внешние наводки, а экранирование земляной скруткой поможет этого избежать. Без этого может наблюдаться такая картина: лента не работает до тех пор, пока не коснёшься рукой сигнального провода.

2.1) If there is a large distance between the tape and the controller (Arduino), i.e. long wires (longer than 50 cm), then the signal wire and the ground should be twisted into a pigtail to protect against interference, since the communication protocol of the tape is quite high-speed (800 kHz), it is strongly affected by external interference, and shielding with an earth twist will help to avoid this. Without this, the following pattern can be observed: the tape does not work until you touch the signal wire with your hand.
[image]

Сделал вот таким образом.

I did it this way.
[image]

DivanX10, 2021-01-27:
Проблема не устранилась. Первые 5 диодов так и горят. Может быть, что это сами диоды повреждены? Ну как так, что с ESP32, что с NodeMCU, все одинаково, только по времени по разному загораются эти первые 5 диодов. В случае ESP32, у них первые 5 диодов загораются через 3-4 дня, в случае NodeMCU только два дня норм было, а теперь снова первые диоды горят через короткий промежуток времени (через 5 - 10 минут).

The problem has not been resolved. The first 5 diodes are still lit. Could it be that the diodes themselves are damaged? Well, how is it that with ESP 32, that with NodeMCU, everything is the same, only in time these first 5 diodes light up differently. In the case of ESP32, their first 5 diodes light up after 3-4 days, in the case of NodeMCU, only two days were normal, and now again the first diodes light up after a short period of time (after 5-10 minutes).

Поменял провода, думал, может в кабеле дело, но кабель тут не причем
I changed the wires, I thought maybe it was the cable, but the cable has nothing to do with it
[image]

Под провода сделал подложку, чтоб исключить вероятность передачи сигнала на диоды.
Under the wires, I made a substrate to eliminate the possibility of transmitting the signal to the diodes.
[image]

DivanX10, 2021-01-29:
Проблема найдена. Всему виной электромагнитные помехи, из-за которых начинают светиться первые 5 диодов (прям как передача энергии индукционным способом). Для проверки я демонтировал ленту и проверял в другом помещении, диоды не загорались. 

The problem was found. It's all the fault of electromagnetic interference, because of which the first 5 diodes begin to glow (just like the transfer of energy by induction). To check, I dismantled the tape and checked in another room, the diodes did not light up.

Обмотал 3-х метровый провод фольгой полностью и сделал заземление, проблему частично удалось устранить, где вместо 5-ти светящихся диодов горят уже 2 диода. 
I wrapped the 3-meter wire with foil completely and made the grounding, the problem was partially eliminated, where instead of 5 glowing diodes, 2 diodes are already burning.
[image]

Заказал витую пару со скруткой в косичку и с фольгой.
I ordered a twisted pair with a twist in a pigtail and with foil.
[image]

Если выше витая пара со скруткой в косичку и с фольгой не поможет, то буду заказывать такой провод
Hyperline Кабель интерфейса RS-485/422 2x(2x22 AWG STP) SFTP (SF/FTP), 120 Ом RS-S/SF2-A-PE
If the above twisted pair with a twist in a pigtail and with foil does not help, then I will order such a wire
Hyperline Interface Cable RS-485/422 2x(2x22 AWG STP) SFTP (SF/FTP), 120 Ом RS-S/SF2-A-PE
[image]

closed 2021-01-29 by DivanX10

bachoo786, 2021-02-08:
Hi @DivanX10  did you manage to resolve the issue?

DivanX10, 2021-02-08:
Yes, I partially defeated the problem. At the moment, the first 5 diodes do not light up. As it turned out later, the diodes were affected by a refrigerator standing behind the wall and a nearby 220v wire. Only a comprehensive solution helped to eliminate the problem, if one thing is removed, then the first 5 diodes of the diodes begin to glow and moreover, this die with 5 diodes did not glow, but glowed under the cabinet.

How did you solve the problem?
1) Installed a 60 x 40 cm screen panel behind the refrigerator, 
2) Stretched the shielding cable and grounded it
[image]

3) Soldered the resistor to 62 ohms
[image]

4) I connected an additional 5 diodes (let's call the LED filter), if you remove them, the first 5 diodes will still be lit, they act as a filter against interference.
[image]
[image]
[image]

5) The wires from the LED filter I have strictly must be in different places, if they are placed next to each other, then the first 5 diodes under the cabinet will light up.

___
Да, я частично победил проблему. На данный момент первые 5-ть диодов не загораются. Как выяснилось позже, что на диоды влияли, стоящий за стеной холодильник и рядом идущий провод на 220v. Только комплексное решение помогло устранить проблему, если что-то одно убрать, то первые 5 диодов диоды начинают светиться и причем, эта плашка с 5-ью диодами не светилась, а светились под шкафом. 

Как решил проблему?
1) Установил за холодильником экранирующую панель 60 х 40 см, 
2) Протянул экранирующий кабель и заземлил
[image]

3) Припаял резистор на 62 Ом
[image]

4) Подключил дополнительные 5 диодов (назовём светодиодный фильтр), если их убрать, то первые 5 диодов все равно будут гореть, они выступают как фильтр от помех.
[image]
[image]
[image]

5) Провода от светодиодного фильтра у меня строго должны быть в разных местах, если их расположить рядом, то загорятся первые 5 диодов под шкафом.

bachoo786, 2021-02-08:
To be fair I haven't solved it yet my issue is only that 1 led glows green occasionally.  I have put my wled strip on my stairs and the cables have been buried in the wall to hide them away in the next room so I only managed to wind the data cable with the ground cable for about 30cm before it connects to the esp8266.

DivanX10, 2021-02-08:
I dismantled the tape from the cabinet for testing and moved it to another room, there were no problems with the glow of the first diodes. Also for the test, I unfolded the tape under the cabinet on the opposite side and one green diode glowed. As I understand it, if there is strong electromagnetic interference, then the diodes will still burn. You need to try the diodes in a place where there are no wires, there is nothing that could affect the diodes. Then, it seems to me that electromagnetic interference works like induction charging for diodes, even with a soldered resistor that should extinguish the current.

------
Я для проверки демонтировал ленту со шкафа и перенес в другую комнату, там не было проблем со свечением первых диодов. Также для теста я развернул ленту под шкафом на противоположную сторону и светился один диод зеленного цвета. Как я понимаю, что если есть сильные электромагнитные помехи, то диоды все равно будут гореть. Вам нужно попробовать диоды в том, месте где нет проводов, нет ничего, что могло бы повлиять на диоды. Потом, как мне кажется, что электромагнитные помехи работают как индукционная зарядка для диодов, даже с припаянным резистором, который должен гасить ток.

IvanUpa, 2022-10-05:
> The problem was solved when I connected the tape as close to the board as possible. I soldered the wires at a distance of 5 cm from the board, and already from the board itself you can pull a wire of any length and I brought the wire to the power supply at a distance of 2 meters. It turns out that the longer the wire from the board itself, the more sensitive the signal cable will be to interference, which leads to such an undesirable effect
> 
> Проблема решилась, когда я подключил ленту максимально близко к плате. Я припаял провода на расстоянии от платы в 5 см, а уже от самой платы можно тянуть провод любой длинны и я подвел провод к блоку питания на расстоянии 2 метра. Получается, чем длиннее провод от самой платы, тем чувствительнее будет сигнальный кабель к помехам, что и приводит к такому нежелательному эффекту

Тоже столкнулся с такой проблемой и читая твои эксперименты натолкнуло на мысль попробовать ферритовые кольца. И у меня это работает, ставлю пару колец на линию от БП - все хорошо, ничего не меняя убираю их - начинаются глюки. Пока это не показать, может проявится и через пару дней, но возможно кому-то поможет.