wled/WLED #1677: Issue with WLED app on iphone, no lights found no matter what

Mrlonch, 2021-01-19:
Hi there,

I am pretty new on this, I am using ESP8266, I managed to flash it and connected to WLED-AP, configured my password and SSID, saved and rebooted, I can see the new Wled wifi connection with the name I gave it, I can connect from my phone to it and control the string of pixels. However, I cannot figure out how to make the WLED app to find my lights, it seems it is not finding the device despite I can see the wifi network for the ESP8266... Am I forgetting something here??

I have read that phone and ESP have to be in the same network, but if I select the Wled wifi one it will directly take me to 4.3.2.1 address to control the lights and cannot move to Wled app unless disconnecting...so not sure how to pair the device with the app... any help will be apprecited!

Thank you in advance!

WoodyLetsCode, 2021-01-19:
Hi,
you have to enter your home wifi SSID and password, so wled can connect to your home network. If your phone and wled are in the same network the app will find the wled device.
If you connect to the wifi that wled has created you can also use the app by manually adding the ip 4.3.2.1 in the wled-app.
Is there a reason why  you use wled's own AP and not connectiing it to your home wifi?

Mrlonch, 2021-01-19:
> Hi,
> you have to enter your home wifi SSID and password, so wled can connect to your home network. If your phone and wled are in the same network the app will find the wled device.
> If you connect to [...]

Hi! Many thanks for the time!!

After flashing ESP8266, I connected to weld-ap and clicked on wifi setup, then I added my home SSID and password. Then I changed from WLED-AP to my own to better identify it.  I saved and rebooted without problems, if I connect to the new name I have to WLED-AP then I can control the lights, but cannot find the esp in my phone Wled app, it keeps searching and searching but nothing, I ha e read that restarting everything several times would work, but not in my case.

nevertheless I do appreciate the work and effort put in wled! Such an amazing tool!!

Mrlonch, 2021-01-19:
I have now tried to re-flash ESP, left everything as it comes except my SSID and password, but still cannot connect through WLED app... neither through wled-lon1.local nor the IP I set in a second attempt... Not sure what I am doing wrong... not that upset as I can still get to control the lights accessing WLED-AP or 4.3.2.1, but what about if I want to set a second device, I would not be able to identify them properly other than assigning WLED-AP1, 2, 3, and so on....

andyshinn, 2021-01-19:
What AP / router are you using at home? Does it listen on 2.4GHz? The ESP8266/ESP32 are 2.4GHz only.

Mrlonch, 2021-01-19:
> What AP / router are you using at home? Does it listen on 2.4GHz? The ESP8266/ESP32 are 2.4GHz only.

Yes, I have just checked it. I get the feeling that I only get to configure the Esp as a hotspot but not really connecting to my home network...Alexa cannot find it neither...

I have watched videos, read wiki and think I followed everything thoroughly but no success

andyshinn, 2021-01-19:
You should only need to configure the SSID and passphrase and it will connect and get a IP address via DHCP. When you click save and connect does it say that it is trying to connect to your network and the WLED-AP goes away?

Mrlonch, 2021-01-20:
> You should only need to configure the SSID and passphrase and it will connect and get a IP address via DHCP. When you click save and connect does it say that it is trying to connect to your network and the WLED-AP goes away?

no, it stays, indeed accessing through WLED-AP seems to be the only way I can connect to Esp. Am I missing anything?

WoodyLetsCode, 2021-01-20:
Can you send a screenshot of your wifi config page?

Mrlonch, 2021-01-20:
> Can you send a screenshot of your wifi config page?

Sure, here it is:
[image]

Hope has nothing to do with the underscore in the SSID...

Many thanks again for your support!

WoodyLetsCode, 2021-01-20:
Have you tried using DHCP instead of setting a static ip?

Mrlonch, 2021-01-20:
> Have you tried using DHCP instead of setting a static ip?

Yep, same result. I do not see any device connecting to my home network that was not there before. Now tried flashing with NodeMCUPy flasher, but it seems the same happens. I can always connect to WLED-AP, but never see the device. not sure what to check next...

Legsmaniac, 2021-01-20:
You haven't set the Gateway?

Mrlonch, 2021-01-20:
Hi there again,

Shame on me! I misread the SSID and it is an 0 instead of an O... Once corrected everything works like a charm! Alexa detected device and can control it, WLED app works smoothly without problem! Now next challenge, to build my 4251 pixel lights matrix!

Thanks @Aircookie and the rest for the amazing work!

closed 2021-01-20 by Mrlonch