wled/WLED #1678: ![IMG_20190927_093729145_HDRlow](https://user-images.githubusercontent.com/16364254/65752936-277ff600-e10e-11e9-806f-43a1f75b7f51.png)

madnat, 2021-01-19:
Hi there @coelner

I found your simple Schematic useful, and am using it to create a custom WLED driver.  My prototype reboots in random time intervals - Interested to learn if you got this to work as you have it drawn, or needed to make modifications?

Thanks

[image]

_Originally posted by @coelner in https://github.com/Aircoookie/WLED/issues/230#issuecomment-535837593_

coelner, 2021-01-19:
Watch this: https://github.com/simap/pixelblaze/tree/master/V3/hardware

I'm not using this schematics anymore. I was too lazy to debug it. Instead I use the ws2812 with 3.3V for small setups.
However the mosfet is maybe to slow for high speed.

The reboot problem seems to an issue with the code itself or powersupply? Some of those cheap esp12 module are internally broken. Just try another one if you use those. 
Try to compile it with debug enable watch the serial outpit. Maybe the watchdog tells something

madnat, 2021-01-19:
Many thanks for your quick reply.  I think it is watchdog related - I noticed with 11.1 the board reboot much more frequently. 
 Also other firmwares (e.g. Tasmota) installed on the same setup run without issue on my prototype.  
 
 However the power supply capacitors in your schematic are different than those specified in the Espressif application notes (10uF and 0.1uf/100000pF) - your schemtic uses 100uF and 10pF.  Was there a reason for choosing those values?

coelner, 2021-01-20:
https://arduino-esp8266.readthedocs.io//latest/boards.html#improved-stability

For the ht7333 are two 10microF capacitor missing (which could be the 10picoF marking, maybe my fault) and the other one is a 200nanoF

madnat, 2021-01-20:
Interesting article - https://arduino-esp8266.readthedocs.io/en/latest/boards.html.

I see they suggest increased stability by adding 1 x 100nF Cap (C1) and an extra resistor (R4) on the RESET pin.

I have the beefed Cap at C1, (100uF), and the second in parallel is beefed up (10uF) - it provides a longer run time, but still resets. 

It would be good to understand what is kicking the Watchdog, and forcing the reset, as the unit is stable with other firmware/sketch loaded.

Does anyone know if WLED produce a record of what caused the reset?

madnat, 2021-01-21:
I have recompiled the sketch and enabled debugging.  Setting board parameters to align closer to my build has helped stability, but I am still experiencing crashes.  

This is what the monitor shows when a crash occurs:
[11:56:48]Fatal exception 0(IllegalInstructionCause):
[11:56:48]epc1=0x402803a0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000ad, depc=0x00000000
[11:56:48]
[11:56:48]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[11:56:48]
[11:56:48]Exception (0):
[11:56:48]epc1=0x402803a0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000000ad depc=0x00000000
[11:56:48]
[11:56:48]>>>stack>>>
[11:56:48]
[11:56:48]ctx: sys
[11:56:48]sp: 3fffeb10 end: 3fffffb0 offset: 0190

I also have a hex dump (attached)
[WLED_11_1.txt](https://github.com/Aircoookie/WLED/files/5849129/WLED_11_1.txt)

coelner, 2021-01-21:
Ok, you need this: https://github.com/me-no-dev/EspExceptionDecoder/tree/1.1.0

and you put the trace in the windows. You need to compile the exact version you flashed onto the device, because the tool needs the right values from the compiled file.

madnat, 2021-01-22:
Before delving much further into code - I thought I'd do some additional research into powering ESP8266 modules.  I found this information very helpful [https://www.youtube.com/watch?v=wf_msvWv1jk](url) where Andreas Spiess takes a deeper look at the HT7333 when used with the ESP8266 module.

He suggests you need a 1000uF cap when using this regulator with ESP8266 (on your schematic a zero is missing :D).  Unfortunatly this is not practical for my use.  

To test this theory, I have tested the module powered by a beefier, off package 3.3V regulated power source and the module has not crashed for over 7 hours.  So I don't think it is a compilation/code issue - it is most likely a power supply problem as you originally suggested.  I am just bemused as to why other firmwares (e.g. Tasmota) does not crash with the sub optimum power supply.

So I will source and test other 3.3V regulators in my preferred package, and see how I can progress from here with my package design.

The lesson learned - don't choose the HT7333 3.3V regulator unless you have space for a 1000uF capacitor.  

I will keep this issue open until I have proved my theory.

Thank you again for your very helpful collaboration on this.

Legsmaniac, 2021-01-22:
> I am just bemused as to why other firmwares (e.g. Tasmota) does not crash with the sub optimum power supply.

To be fair, neither does WLED for many of us, myself included.  I run several ESP8266's running WLED with a 12V 3A/5A/20A PSU and a buck convertor to reduce the 12V down to 5V powering both the ESP and the LEDs and have no problems whatsoever.
Why 12V PSUs and not 5V?  Because in some situations, the distance from my PSU to my lights is great, resulting in voltage drop due to resistance of the wire.  So by using a 12V PSU, the resultant voltage could be (say) 8V at the point of installation of my lights and a Buck Convertor controls the voltage at 5V constant.  Whereas if I had started with a 5V PSU, the resultant voltage could be 3V at my lights and not enough to power my system. 

I realise there are a good few people on here having problems with reboots and dropping Wi-Fi but I believe there are a great many more using WLED with no problems so you are correct in saying it's not a code issue (although it "could" be a compilation issue if not compiled correctly).  As for other firmwares (eg Tasmota) they are probably used with short supply leads compared to most WLED setups?

coelner, 2021-01-22:
I use the ht7333 in solar powered devices and I had some of them. And I reduce the tx power to the least I can , so the peak is less high.

madnat, 2021-01-22:
> To be fair, neither does WLED for many of us, myself included. I run several ESP8266's running WLED with a 12V 3A/5A/20A PSU and a buck convertor to reduce the 12V down to 5V powering both the ESP and the LEDs and have no problems whatsoever.

Hi @Legsmaniac  and thanks for contributing to this topic.

For the avoidance of any doubt - I'm a huge fan of WLED -  its robustness and quality I do not question.  Just perplexing to me  how the ESP8266 can decide to crash running some software and not with others.  

I have many other WLED implementations based on the usual pre-packaged varieties of ESP8266, and really very happy with it -delighted infact, which is why I decided to step into building a custom ESP8266 package for a plug and play solution.

My plug and play solution is actually a 12V supplied unit, with a 5A 5V buck converter to drive LED's and level shifter, the 3.3 is only there to drive the ESP8266.  If you have significant experience of creating a custom ESP8266 package from a 5V power source - please do share it, as I'm about to emark on trialling various 3.3 regulators in a SOT89-3 package, I'd not have to reinvent what is already known to the community.

My proto board has space for 2 capacitors (0805 and 0603) between the 3.3V regulator and the 3.3 power pin of the ESP8266, so and suggestions of caps to deliver a stable package would be awesome.  Currently they are occupied with 100uF and 0.1uF units.

madnat, 2021-01-22:
> I use the ht7333 in solar powered devices and I had some of them. And I reduce the tx power to the least I can , so the peak is less high.
> 
> But you can choose the ht7833. Or ht7335,... ( https://arduinodiy.wordpress.com/2020/01/18/very-deepsleep-and-energy-saving-on-esp8266/ )
> 
> WLED squeeze the maximum out of these things. Tasmota consider energy savings

Ahh - good tip - WiFi performance is not that critical for my use case, so I can try tweaking this.  Would you mind pointing me in the right direction to adjust Tx power in the Sketch?  Thank you again.

coelner, 2021-01-22:
I use this part of sketch to do this, but remember, I used deepsleep and the rtc ram to save data. You need to adapt this, but atleast you can see the logical flow from it. 
```
bool wifi_connect() {
  bool rtcValid = false;
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);
  WiFi.forceSleepWake();
  delay(10);
  WiFi.mode(WIFI_STA);
  yield();
  // Disable the WiFi persistence.  The ESP8266 will not load and save WiFi settings in the flash memory.
  // http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html#persistent
  WiFi.persistent (false);
  WiFi.config(ip, gateway, subnet, dns);
  // Connect to WiFi access point.
  if( rtcValid ) {
    // The RTC data was good, make a quick connection
    if (rtcData.txPow > 0 ) {
      rtcData.txPow=rtcData.txPow-1;
      WiFi.setOutputPower(rtcData.txPow);
      yield();
    }
    WiFi.begin( WLAN_SSID, WLAN_PASS, rtcData.channel, rtcData.bssid, true );
      Serial.println("RTC WiFi used");
      Serial.println("\tCurrent Powerlevel: \t" + String(rtcData.txPow*4) + "dB");
  }
  else {
    //raise outputPower to max
    WiFi.setOutputPower(10);
    rtcData.txPow = 10;
    // The RTC data was not valid, so make a regular connection
    WiFi.begin( WLAN_SSID, WLAN_PASS );
    yield();
    delay(10);
  }
  Serial.print("WiFi started");
  int i = 0;
  uint64_t sTime = millis();
  while (WiFi.status() != WL_CONNECTED) {
    delay(200);
    i++;
    yield();
    Serial.print(".");
    if ((millis() - sTime) > WIFI_CONNECT_TIMEOUT * 1000)  // wifi connection lasts too ling, retry
    {
      Serial.println("WiFi connect timeout!");
      //raise transmit power for next retry
      rtcData.txPow=rtcData.txPow+1;
      return false;
    }
    //every two second raise the TX power 
    if (i%4 == 0)
    {
      rtcData.txPow = rtcData.txPow+1;
      WiFi.setOutputPower(rtcData.txPow);
      yield();
      Serial.println("\tWiFi TX Power raised: \t" + String(rtcData.txPow) + "counter");
    }
    yield();
  }
  //ToDo stupid TX power correlation --needs improvement
  long rssi = WiFi.RSSI();
  //rtcData.txPow = int((72+rssi)/4)+rtcData.txPow;
  if (rssi > -60 ) 
  {
    rtcData.txPow = 0;
  }
    Serial.println("\n\r\tWiFi Connect Time: \t" + String(float((millis() - sTime)/1000)) + " s");
    Serial.println("\tWiFi RSSI: \t\t" + String(rssi) + " dB");
    Serial.println("\tWiFi next TX Power: \t" + String(nv->rtcData.txPow) + " counter");
  // Write current connection info back to RTC
  Serial.println("\tIP address: \t\t" + WiFi.localIP().toString() );
  return true;
}
```
Absent from learning things, why do you not use an recommend board? IMHO this community does put a lot effort into these boards (https://github.com/Aircoookie/WLED/wiki/Compatible-hardware) so why bothering with designing a new instead extend an existing one to your needs?

If you want to design a new pcb, you should have an oscilloscope, otherwise it will be a hassle.

Legsmaniac, 2021-01-22:
> If you have significant experience of creating a custom ESP8266 package from a 5V power source - please do share it

It's simple really.  I use boards such as the [NodeMCU 8266](https://www.az-delivery.uk/products/nodemcu-lua-amica-v2-modul-mit-esp8266-12e-unverloetet?_pos=28&_sid=227f277fa&_ss=r) or the [Wemos D1 Mini](https://www.az-delivery.uk/products/d1-mini-v3?_pos=31&_sid=227f277fa&_ss=r) which have onboard regulator and USB drivers so I don't have to bother faffing with an FDTI to program them or additional 3.3V regulators.  I simply plug the boards directly into my PC to upload WLED (or whatever program I'm using them for at the time) and the boards have both 5v and 3.3V pins and GND so I use the same 5V from the Buck that is powering the LEDs.

madnat, 2021-01-22:
Yes - I did consider using stock modules  - and in hindsight should have stuck to that option.   The  board created by the community (led by @srg74) is clearly excellent, but is unfortunatly too big for my application.

Programming via FTDI really isn't that big of a deal.

I also saw these articles and thought it would be relatively easy to setup a more elegant solution for my custom use.  Furthermore, I am interested in learning how this stuff works and picking up knowledge from others like you.

It wasnt my intention to design a completely new PCB, but more take a working schematic and apply it to a custom layout.  Saying all that, I am confident I'll get to a successful outcome, with either the right combo regulator/caps or a custom build firmware that is less demanding.  In either case I'm happy to share my solution.

Legsmaniac, 2021-01-22:
> Programming via FTDI really isn't that big of a deal.

No, I do agree.  I have done a few myself.
But the small size of the Wemos D1 Mini really isn't that much bigger than (for example) the ESP-01 but I guess it does depend on what you want to do with it, fair enough.  Mind you, by the time you've added a 3.3V reg, the combined size of the reg and the 01 would be the same if not a bit bigger the the Wemos.   I do prefer it being all on the one board but that's just my personal preference I guess.  :-)

relax81, 2021-01-23:
The HT7333 is limited to 250mA max if the datasheet I found  is correct - that's not enough to run it stable from my experience.  On Reddit and other places there are whole discussions about "some" Wemos D1 Mini Boards for example that crash if the manufacturers use too weak voltage regulators.  Especially during WiFi search and such operations it can spike up to 300/400mA for a very short moment. 

On my boards i use a 10µF & 100nF cap close to the VCC pin of the ESP12-S and either the SPX1117M3-L-3-3/TR or the XC6210 B332MR Voltage Regulator (with the caps that are recommended in their datasheets). They're both rated up to 800mA if i remember correctly.

madnat, 2021-01-23:
I'm going to try a selection and report back what I find... these are available on Mouser, that will fit the parameters @relax81 mentions (very helpful by the way):
AZ1117CR2 [https://www.mouser.co.uk/ProductDetail/621-AZ1117CR2ADJTRG1/](url)
AP7365-33YRG [https://www.mouser.co.uk/ProductDetail/621-AP7365-33YRG-13/](url)
XC6214P332PR [https://www.mouser.co.uk/ProductDetail/865-XC6214P332PR-G/](url)

Others are available - but these all have the same pad/pin layout as my current proto board, which limited my choices.

To add candidates for custom builds, reading other blogs and reviews, many of the current ESP32 boards (I assume these are more power hungry than ESP8266?), use ME6211, AMS1117, RT9080, EA3036 and AP2112.  However, I could only find a reliable source for AP2112 in the UK (but not available in my package required spec).

madnat, 2021-01-28:
I have received my new 3.3V regulators from Mouser today.  I ordered the wrong version of the AZ1117CR2 (the link is for an adustable version - not 3.3V) - but fortunatley the right version of the XC6214.  Currently soak testing with this 600mA regulator fitted - seems to be going strong, will see if there are any crashes overnight.

madnat, 2021-02-01:
All tests have gone well with the new regulator.

If you'd like to see my finished project please watch my video here:
WLED for a Pinball Machine [https://www.youtube.com/watch?v=SbzJ4uGXotk&t=357s](url)

If you'd like to buy one - please contact me.

closed 2021-02-01 by madnat

clove0, 2023-01-09:
Thought I'd add my voice in here in case it helps someone...

My board is "custom" PCB with an ESP-12F. The ESP is powered via a 5V power supply fed through a BA033CC0FP-E2 3.3V regulator. The regulator has a max current output of 1A...so current shouldn't be an issue. 

I was running WLED on two identical boards which were having the same issue of "crashing". I tried adding a 2200µF to both the 5V and 3.3V rails but was still having issues. I made sure that WLED was updated to the latest version (it was). 

What solved the problem of "crashing" was the cheap NO pushbutton I was using. After removing the one connected to the reset pin and soldering in a jumper to tie it high the problem went away. 

For inquiring minds, the switch model number is TL3305AF260QG.