wled/WLED #1680: Macros have moved (to where?)

velkrosmaak, 2021-01-20:
I'm trying to configure a button to randomise the current WLED effect, but it appears the documentation refers to a different version where I'm able to configure 'Advanced Macros' in the UI. See screenshot here: https://github.com/Aircoookie/WLED/wiki/Macros

How can I define the Macro ID which will perform the FX=~ action? Thanks!

Doyle4, 2021-01-20:
Config > Time & Macros

velkrosmaak, 2021-01-20:
The UI under Config > Time & Macros does not have the same settings as in
the linked documentation. Where I would expect to find Macros it says
"Macros have moved"

YvanLeTerrible, 2021-01-20:
Macro has moved...into spaaaace
I have same issue, tried release 11.0 and 11.1, both UI look missing the "advanced macro" link/menu button : no area for filling API commands
New preset layout was really good but I have to roll back to release 10.2 because i really use having  available macro ( T=2 on/off toggle on single press, and also preset cycling for double press on external manual physical button)

velkrosmaak, 2021-01-20:
It looks like you might need to define and configure these custom macros
via the API now, so for speed have also reverted to 10.2 and was successful
setting this up in the UI.

closed 2021-01-20 by velkrosmaak

Norcom0s, 2021-02-17:
You have the enter API commands in a  new Preset and call this Preset via the Timers and Macros tab.
To perform a switch led off comand at a defined time: 
you create a new preset, reboot your wled via the info screen(otherwise you will not see any API Commands in the preset details)
Now you select your new preset and you'll see the saved API Commands and the preset no.(remind this later) 
Here you can change the  "on":true to "on":false.
Press save and go to Config->Time and Macros. 
Ensure that the time and Date is correct and scroll down to time controlled presets. 
Choose one timer, type in your switching time and enter the preset No. of your previously created preset.
Press Save 
Thats it!
For button action use the area above the time controlled presets section. 
Simply enter the preset No. you want to execute under the defined condition.
Have Fun!

DonGar, 2022-09-28:
With the new system, is it still possible to Toggle the lights on an off via a short button press?

blazoncek, 2022-09-29:
The *new* system is now 2 years old. 😄 
Yes, it is possible to do that (it is the default behaviour) and much more.

Sebbb, 2023-04-27:
> The _new_ system is now 2 years old. 😄 Yes, it is possible to do that (it is the default behaviour) and much more.

I'm struggling to find out how to solve a simple (?) on/off problem.

Button 0 does toggle on/off per default. But it also changes colors when pressing longer and and resets WLED to the default settings when pressed toooo long. So I would like to use Button 1 for that.

So I have figured out that a preset {"on":"true"} toggles my LED, so I assigned that to my button 1. But is there a way to change the light instantly when pressing, not when releasing the button? I tried setting the same preset for short and long press, that doesn't do the trick....

Can anyone give me a hint?

Thanks!