wled/WLED #1681: PIR an Wemos D1 WLED

Sirbuschi2003, 2021-01-20:
Good day,

there is a possibility to connect a PIR to the Wemos D1 to switch the WLEDs on and off.
Function would be if Time 8 p.m. -10 a.m. it should turn on when there is movement.
Of course, dimmed and so on selectable.

Otherwise, the stripe should still be controllable via app. APP takes precedence over PIR etc.

that would be possible. The ulterior motive is that I want to operate everything on my bed as night light and ambient lighting.
THANKS

Legsmaniac, 2021-01-20:
This is already possible.  Have a look in the Usermods folder, there are two folders beginning with PIR.  You would need to add one of these files to your WLED and recompile.

Sirbuschi2003, 2021-01-26:
I can't do it with the instructions. Can somebody create a finished image which one then only has to bottle or prepare the files for the compiler, something I'm doing wrong

Sirbuschi2003, 2021-01-26:
ich habe alles nach anleitung eingestellt aber es geht nicht.

Rushmed, 2021-07-15:
Currently I am using precompiled 0.13.0-b2 on a Wemos mini with an AM312 PIR attached.
Motion detection works, LEDs are automatically switched on an off one minute later usin nightlight in the triggered preset.
Thats great so far. But it`s doing this all day long. So I am also intrersted in the ability to switch on/off motion detection according to daytime or even better to sunrise and sunset as @Sirbuschi2003 requested.
This functionaltiy is not included in the usermods.
Is there a way using presets? Tried saving different presets each sunrise and sunset to the preset that is triggered by PIR but the off preset was not working properly.

blazoncek, 2021-10-05:
Pre-built binaries with usermods are available in @srg74 Wemos shield repository for your convenience.

closed 2021-10-05 by blazoncek