wled/WLED #1688: Add Bounce Option to all relevant effects

qualar, 2021-01-23:
Each effect currently has a speed and intensity slider.  It would be nice to also add a bounce toggle.  When enabled that particular effect when complete would reverse backward and forward. eg a chase effect that goes from left to right would chase left to right then bounce back right to left and then repeat.

Fredyy90, 2021-01-23:
this would be a nice addition to always have a "perfect loop"

Model-Man, 2021-01-23:
I have already requested a similar option some time ago. Not sure if Aircoookie is considering it. 
I suggested having a auto reverse checkbox box on the main effects screen with the addition of a timer slider (1-30 sec) that would reverse the sequence at a predetermined time, say 20 seconds. Hopefully this feature could then be saved as a preset. 
Auto reverse definitely adds additional interest to any effect rather than just running in one direction.

blazoncek, 2023-08-05:
FYI this would require rewriting each and every effect (that does not already include cycling).
So IMO it will never happen.