wled/WLED #1691: How to get actual state or brightness

nicedevil007, 2021-01-25:
If I just want to check the actual state of the LEDs brightness or the "on/off" state, how can I achieve this?

My prefered solution is MQTT. An empty message to the topic wouldn't send anything :/

andyshinn, 2021-01-25:
Have you tried the `v` topic such as `wled/298232/v`? It should have the status in XML format where `ac` is the brightness.

nicedevil007, 2021-01-25:
mhhh yes but won't return anything, I assume your `298232` must be replaced by my string I setup in the mqtt tab.
so f.e. I have a mqtt topic 

`wled/kitchenbulb` i have to send a message to the topic `wled/kitchenbulb/v` right?

andyshinn, 2021-01-25:
Correct. It is not giving back a XML string on change?

closed 2021-06-09 by stale[bot]