wled/WLED #1692: Feature Request: On/Off at Sunrise, Sunset

DoctorWiz, 2021-01-25:
Would like to see WLED calculate sunrise and sunset times and be able to turn on and off (or run preset) based on those times, better yet allow an offset.  You already store & calculate time zone, offset from GMT, savings time, etc.  Just need to add Longitude and Latitude for the calculation.  Plenty of various algorithms and code examples thru a google search.  Possible way to implement this might be to put "sr" or "ss" in the hour field, minute field would be offset (and allow for negatives).  Or via some other page/fields if you think that's better.  Example: Preset #X (On) at sunset + 15 minutes, Preset #Y (Off) at sunrise - 15 minutes.

Thanks for all the amazing work you have already done on this project!
🧙‍♂️ Doctor Wizard

blazoncek, 2021-03-11:
Made a PR #1796 for that.

qai-retired, 2021-05-20:
Keyboard pop-up doesn't have a negative key to allow entering negative long/ lat coordinates

closed 2021-10-05 by blazoncek