wled/WLED #1693: Turns on after a while

mazmazen, 2021-01-25:
In my garage I have installed 3x300 leds 5 meters. All 3 goes on the same pin on the wemos D1 mini 8266. The issue is that if I turns off the leds in the app after a while they automatic turns on again but only about 30 leds on each strip. The first led is yellow the rest is White. The strips are powered from a 26a psu and in the app the the strips are configured to use 8a.
It is flashed with the newest firmware.

[image]

derekelijah, 2021-01-25:
Mines been doing this as well but it's sometimes purple or blue.

sansillusion, 2021-01-26:
That is interference on the data wire, one good way to prevent this is to
use a relay on the LEDs power line so (with wled relay mod) the LEDs power
is off when wled is off.

mike2nl, 2021-01-29:
Or use shielded cables after the level shifter. If you don't use a level shifter i woud adivse it. Coud be the 74HCT125 or an ordinary RGB led directly on the chip output. And from there you use a shielded cable to the first LED strip.

- So as @sansillusion mentioned it coud be an interference on the data line. Mostly that is the issue. Or you have some neighbours who playing with your WLED stuff. 
- Is your WLED device connected to the router so that the device can be reached from outside? If so it not a good idea to do it in that way.

hannesbe, 2021-02-16:
I'm having a similar issue, but only after updating to 0.11.1. If the issue is interference on the data wire, this would also have occurred on 0.10.2, or is this too short-sighted?

[image]
[image]

Legsmaniac, 2021-02-16:
> Use shielded cables after the level shifter. If you don't use/have a level shifter i woud adivse it.

That's exactly what I have done with my Christmas roof lights.  The distance from the WLED device to the first LED (at the peak of the apex of the roof) is some 15 - 20 feet away, hence it was important, I thought, to use shielded cable and a level shifter.  And it's certainly helped.  I get (got, since Christmas is over for another year) no problems at all.

SpudGB, 2021-02-16:
Mine are doing the same on 11.1

Its not interference or cable length as 10.2 doesn't do it at all. Mine intermittently turn on about 7 leds (alternating white, blank, white etc.) over the first 12 leds. 

Its almost as though wled is doing a reboot but not defaulting to orange afterwards and staying off. Turn the leds on/off via Alexa, button, or app and it goes away but comes back after an indeterminate time. All other functions work without fault afaict.

nonsintetic, 2021-04-11:
I have a similar issue WITH a relay. The relay is off (i can hear it turn on when i turn it on), but 2 leds that are on a farther away strip always turn on.
I think on top of noise causing random pixels to appear, I'm getting fantom power through capacitive coupling of the relay. Really weird, not sure how to fix it.
So far I'm just turning the leds on for a split second and off again at 8AM every day to mitigate it. Would be cool to have a setting where WLED tells the leds to be OFF once every X hours or minutes if they're supposed to be off.