wled/WLED #1694: Color settings

baerni, 2021-01-25:
Hi. 3 colors can be set in the UI. Is it possible to set one of them to a color from the palette? Or is it possible to set a random change of one color with a slight transition? I would like to set color 1 as white and color 2 as random / rainbow / palette color.

Doyle4, 2021-01-25:
My work around is to create segments, give each segment an effect

baerni, 2021-01-26:
I don't know if you understood what I mean. Example: We have a running effect, we can set color 1 and color 2. For example we will set color 1 as white and color 2 as blue. I would like color 2 to be not only blue, but that I could set color 2 as rainbow / palette color / random with a smooth transition.

Doyle4, 2021-01-26:
I see what you're saying.
That be a good add on.