wled/WLED #1695: HA native integration behaves differently than the Web UI

denes44, 2021-01-25:
Hi,

I'm playing with WLED for the first time, and the problem I have, is that the controls in the Web UI behaves differently than the controls in Home Assistant with the native integration.
Using the Web UI, the brightness in HA is always 255, and if I modify the brightness in the Web UI, it changes the RGB values of the current color. That way I can adjust the separate white channel's brightness and the color's brightness independently.
Is there a way to achieve the same behavior in HA as in the Web UI?

Another issue, is that when I click on the center of the color wheel in the Web UI, the LEDs become white, however, if I do the same in HA, the LEDs will be switched off.
The attributes in HA is the same for both ways:
brightness: 255
hs_color: 0, 0
rgb_color: 255, 255, 255
xy_color: 0.323, 0.329
So if I click on the white in Home Assistant, the integration turns off the LEDs for whatever reason, although it shouldn't

Legsmaniac, 2021-01-25:
I do believe that this question is best asked in the HA forum as the add-on for WLED for HA is written independently and specifically for HA.

That said, I use HA and the latest WLED and I cannot reproduce your issue. Clicking the middle for white gives me white and doesn't turn my LEDs off.

denes44, 2021-01-26:
Oh, okay, I thought the HA integration is managed also by WLED.

I found the cause of the white turn off problem, this is the relevant source code of the HA integration source code:
```
# On a RGBW strip, when the color is pure white, disable the RGB LEDs in
# WLED by setting RGB to 0,0,0
if data[ATTR_COLOR_PRIMARY] == (255, 255, 255):
   data[ATTR_COLOR_PRIMARY] = (0, 0, 0)
```

Legsmaniac, 2021-01-27:
I'm no coder, no expert but I'm pretty sure that the above code you picked out relates to RGBW LED strips and so would be correct.  In other words, when selecting only pure white, it is turning off the Red, Green and Blue leaving the White turned on.  It doesn't apply to addressable LEDs such as WS2812 which uses RG & B to create White.

At a guess, if you are using RGB strips and have told WLED you are using RGBW then yes, I expect you could well get this problem.  What kind of strips are you using?

denes44, 2021-01-27:
Well, I'm using SK6812 addressable RGBW strip :)
I got this answer from the codeowner of the integration:
`Yes, this feature has been implemented by request and in cooperation with the author of WLED (aircoookie).`
However, this doesn't function properly, because when it turns of the RGB part of the LEDs, it doesn't turn on the white part of the LEDs, so the strip became dark... This doesn't make any sense.

And my other issue is, that I bought a led strip with warm white LEDs, so I don't want these white LEDs to take over the white of the RGB LEDs :)
I bought this strip, so I can use it in RGB mode with Philips Ambilight, and in white mode when I just want to use some warm white for mood lighting, and I don't want to use the ugly cool whites of the RGB LEDs. :)
So basically I want the two modes (RGB @ warm white) to be completely separate. WLED has a settings for this called 'Auto-calculate white channel from RGB modes', and it works quite good in the UI, I have separate controls for the colors, and for the white, and the does not affect each other's brightness.

However this doesn't work with the native HA integration written specifically for WLED.

The brightness thing is just one thing, it doesn't work as I want to, however the way it works does not coherent with the settings in WLED.

But the handling of the white color is completely wrong, it assumes that I want to use the white LEDs rather than the white of the RGB LEDs, but it just turns off the RGB LEDs, and do nothing with the white LEDs, so I can't select white in the integration, because if I do that, the strip became dark...

The codeowner however doesn't aggree with me, that this behavior is wrong.

Legsmaniac, 2021-01-27:
> I don't want to use the ugly cool whites of the RGB LEDs

I use RGB and I don't get/use "ugly" whites. LOL
I set my "whites" to be warm white, antique white etc. which of course is controlled by the amount of RGB.
Warm White for example being #FDF4DC  (253, 244, 220) and Antique White #FAEBD7 (250,235,215)
My view being it gives greater control over the whites and works very well, rather than a fixed white (or warm white).
Just my opinion of course but each to their own.  :-)

> The codeowner however doesn't aggree with me, that this behavior is wrong.

Which is exactly what I said.  I'm inclined to agree with the codeowner I'm afraid.

Guess we'll just have to wait for @Aircoookie or one of the other techies to pop along and see what they say.
I can't help wondering if you may have a setting wrong somewhere but I don't have any RGBW to test.

denes44, 2021-06-02:
Just for the record, if anybody sees this thread in the future, ESPHome can handle RGBW lights properly with Home Assistant.

closed 2022-04-16 by stale[bot]