wled/WLED #1697: support for sk9822 (close to apa102)

damnms, 2021-01-26:
**Is your feature request related to a problem? Please describe.**
i have a sk9822 stripe which does not work as expected - some lights are blinking, some not, seems like random, because apa102 can be compatible programmed but seems like its not in WLED 

**Describe the solution you'd like**
make apa102 compatible to apa102 and sk9822

**Describe alternatives you've considered**
add support for sk9822

**Additional context**
https://cpldcpu.wordpress.com/2016/12/13/sk9822-a-clone-of-the-apa102/#:~:text=A%20protocol%20that%20is%20compatible,0x00%3E%20

I was trying to figure out how to use sk9822 in fastled in WLED but i am somehow unable to find that code location 
https://forum.makerforums.info/t/using-fastled-with-an-esp32-and-apa102-so-i-saw-this-issue/62540
i did nowhere find anything like FastLED.addLeds() in WLED, so my assumption is that fastled is not used at all?

damnms, 2021-01-27:
guess this should be somehow transformed to a bug report because... seems like sk9822 should be supported, when i take a look at https://github.com/Aircoookie/WLED/issues/684
but mine are working very very badly, so something strange is going on :/

feindsender, 2021-01-27:
Hi,
is your Neopixelbus library up to date ? SK9822 are supported since March 2020.
https://github.com/Makuna/NeoPixelBus/commit/d3014985025997845cbdc4cb9689f3d88c83b2bc

You can read about the problems before here:
https://wled.discourse.group/t/leds-flickering-while-using-brightness/142

greetings
Maik

damnms, 2021-01-27:
its working now... it was probably just a configuration issue. i did not notice that the mA are limited and the rgb order was wrong.
i raised the oomph to 20.000 and rgb to rgb and now its working

closed 2021-01-27 by damnms