wled/WLED #1698: Add possibility to tie WLED Button to MQTT message(s)

careyer, 2021-01-27:
I am in the process to build a "notification lamp" based on WLED. Depending on a variety of differnt events that happend in the houshold (e.g. washing machine finished, dishwasher ready, new snailmail in mailbox, etc.) the lamp shall light up in differnt colors and if no event is due stay off.

Now it would be nice to have Button tied to WLED that you can press to confirm that you have perceived the notification and turns the light off. Prett much a buzzer that you can hit to confirm you have seen the notification.
The buttonpress shall then send a specific MQTT message to the configured MQTT broker.

This would be awesome and would take confirming notifications with WLED much much simpler.

Thank you very much!
Keep up the great project!

Best regards
careyer

sife14, 2021-03-17:
This can be done with a usermod. You can modify and adapt something like this [Usermod](https://github.com/sife14/WLED/blob/master/usermods/mqtt_button_input/mqtt_button_input.h). 

Turning off the light with a hardware button is already supported by WLED

blazoncek, 2021-10-05:
Buttons can publish MQTT messages in 0.13.
Closing.

closed 2021-10-05 by blazoncek

janbernhardt, 2022-04-09:
What needs to be done to publish a button action via MQTT in 0.13?
I enabled Button 0 on my LED configuration and selected type PIR sensor. But on my MQTT broker I do not see any button status messages. My light turns on and off, when my sensor detects motion, so the general configuration seems to be working.
Is it also possible to get only the button message on the MQTT message without any light effect changes?

janbernhardt, 2022-04-09:
I had to enable "Publish on button press" as well. Now I'm getting my motion on/off message as well.

janbernhardt, 2022-04-09:
I was able to solve my lights on/off effect during motions, by defining two custom API presets in my macros. Turning my night ligh on or off.

    {"nl": {"on":true,"dur":10,"mode":1}}

    {"nl": {"on":off,"dur":10,"mode":1}}