wled/WLED #1700: ZigBee integration

b3-4r, 2021-01-28:
With ZigBee-based colour changing bulbs becoming cheaper and cheaper (like IKEA's Trådfri, Lidl recently started their own line of smart bulbs, etc.), I've been thinking it would be super useful to be able to sync general room moodlighting to WLED-based features to be able to include background lighting in scenes. This should be relatively easily doable using a home automation system, but to set one up just for lighting control seems a bit overkill when one extra ESP would do the trick. Same goes for using official (online account requiring, nonfree) gateways.

Would this be easily achievable?

neckyaan, 2021-08-20:
Bind zig device to mqtt service through zig2mqtt, it should be controllable

AskAlice, 2022-10-08:
The TTGO T-Zigbee board has an ESP32-C3 and a TLSR8258, which when flashed with the proper firmware provided by ttgo, can be communicated with from the ESP32 to accept and receive zigbee messages. This should create an interface where WLED could acquire zigbee support. The whole point of zigbee is to create a meshnet that is unaffected by wifi interference, so while I understand why you'd suggest using zigbee2mqtt, the implementation that I've suggested would enable wled to work regardless of the shitty wifi antennas on far too many ESP32/8266 boards.