wled/WLED #1701: Brightness fader

maximbeliveau, 2021-01-28:
The brightness fader is not changing the intensity when dragged. It changes Just when you click on the fader at a different spot 

Running on the most recent binary version on esp01

Thank you for your help!

mike2nl, 2021-01-29:
@maximbeliveau 
- That is not a bug. It was coded in that way.
- Possible you can add a PR with your code to change it.
- Otherwise you have to wait if someone is willing to change this or @Aircoookie will do this.

Aircoookie, 2021-02-05:
Hi and thank you for the report! If you drag the slider and then release it, does it update the brightness? If not, what is your device and browser?

If you mean that the brightness is not updated in real time _while_ the slider is dragged, that is, as @mike2nl pointed out, indeed a design choice to increase stabiliy.

closed 2021-06-07 by Aircoookie