wled/WLED #1706: Support for Raspberry Pi Pico?

Suspect-byte, 2021-01-29:
Hey, will there be support for the Raspberry Pi Pico?
Thanks in advance for answering.

Doyle4, 2021-01-30:
I have a couple of the Pico's.

To be honest, for LED's, ESP's are best to use as they have built in Wifi and are same price there abouts.
For a pico to have wifi you need to hook it up to a Raspberry Pi, or a wifi board, but the speed is slow.

Picos are a micro controller over a development board.

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Visit this channel for raspberry pi pico projects..

panchalnikunj, 2021-02-13:
[``](https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ)

panchalnikunj, 2021-02-13:
[YouTube](https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ)

Visit this channel for raspberry pi pico projects.

Aircoookie, 2022-04-16:
Closing for now since I don't think support is feasible without built-in WiFi. To add wifi, it is proposed to add an ESP which is rather ironic: https://www.tomshardware.com/how-to/get-wi-fi-internet-on-raspberry-pi-pico

closed 2022-04-16 by Aircoookie

TimBroddin, 2022-06-30:
Raspberry Pi has released the Pico W, with built-in wifi: https://www.raspberrypi.com/news/raspberry-pi-pico-w-your-6-iot-platform/ 🥳

blazoncek, 2022-06-30:
> Raspberry Pi has released the Pico W, with built-in wifi: https://www.raspberrypi.com/news/raspberry-pi-pico-w-your-6-iot-platform/ 🥳

You would need to port all of the libraries.

liftoff-sr, 2022-06-30:
This should now be reopened?  Are 2 mbytes of flash going to be enough with the WIFI code presumably taking up some of that?

blazoncek, 2022-06-30:
Have you ported all of the libraries? 😲

jmerc77, 2022-12-13:
may I suggest to split the code in to 2 layers?
for example firmware on flash, the rest (such as effects) on an SD card?

blazoncek, 2022-12-13:
> may I suggest to split the code in to 2 layers? for example firmware on flash, the rest (such as effects) on an SD card?

An interesting idea. Can you show us how, please?

softhack007, 2022-12-13:
> This should now be reopened?

I think its still not feasible. WLED is strongly depending on libraries from espressif, which do not exist on RP2040.

jmerc77, 2022-12-14:
in #2704 there was mention of some replacement libraries that can do the same tasks as the esp32 ones:

@Cryptophunk :

> ESPAsyncWebServer
> ESPAsyncTCP
> ESPAsyncUDP
> IRRemoteESP8266
> LittleFS
> 
> These wouldn't be required as these are hardware specific libraries that have alternatives for RPI.
> 
> There's going to reach a point where you can only fit so much on an ESP32.

Cryptophunk, 2022-12-14:
@jmerc77 The library you are probably looking for to port WLED to rpi would be FastLED as this is the library to send data to the lights. I also believe that the server is built with NPM then minified after so it can fit on the ESP. I haven't looked into the sound reactive stuff much so I'm not sure if there are seperate libraries for that or if it is already supported in FastLED. 

The first library I mentioned in the quote is to enable web server capabilities for ESP. The second and third are for UDP and TCP sync and the fourth is to allow saving onto an ESP chip. These libraries you can completely ignore when porting to rpi.

blazoncek, 2022-12-14:
Unfortunately none of that is true.
WLED uses NeoPixelBus, not FastLED. ESPAsyncWebServer is crucial for WLED as it defines what it is. LittleFS would need a replacement FS driver in Arduino architecture. If you do not want external connectivity then you *can* ignore ESPAsyncTCP and ESPAsyncUDP.

Cryptophunk, 2022-12-14:
What I meant to say is LedFX not FastLed. Not sure if you would need to replace FS if you're not building it on an Arduino architecture. It's also kind of obvious that WLED runs on a webserver which would be much more easy to expand upon on rpi than an ESP device. There's not just one way to rebuild this for another architecture.

blazoncek, 2022-12-14:
What is LedFX? A separate unrelated program. Yes it can output data WLED will understand.
FS access is mandatory for storing configuration and presets. Using Arduino platform wrapper. An finally WLED was written using Arduino platform in PIO, RPi uses it's own platform which is incompatible with Arduino.
Of course you can always write from scratch.

Sojourneer, 2024-05-09:
The RP2040 has PIO, which supports buffered clocked bitbanging, and easily implements the WS2811 protocol.   The SDK contains a WS2811 driver as an example.

blazoncek, 2024-05-09:
> The RP2040 has PIO, which supports buffered clocked bitbanging, and easily implements the WS2811 protocol. The SDK contains a WS2811 driver as an example.

We'll be most grateful when you port all libraries to RPI.
Perhaps read all of the posts above.

Sojourneer, 2024-05-09:
> > The RP2040 has PIO, which supports buffered clocked bitbanging, and easily implements the WS2811 protocol. The SDK contains a WS2811 driver as an example.
> 
> We'll be most grateful when you port all libraries to RPI. Perhaps read all of the posts above.

8 hardware peripherals for WS2811 on a cheap chip is quite attractive, but not for my small installations.  I'm using the Pico as a WS2812B test driver, and thinking about making it a WS2811 USB device so I can run it from Python.  We all have our priorities.

blazoncek, 2024-05-09:
FYI ESP32 can theoretically do 2x 24 outputs using I2S + 8 outputs RMT, I've tested 1x 16 outputs with I2S + 8 outputs RMT. HW assisted. On the cheap.

Sojourneer, 2024-05-10:
> FYI ESP32 can theoretically do 2x 24 outputs using I2S + 8 outputs RMT, I've tested 1x 16 outputs with I2S + 8 outputs RMT. HW assisted. On the cheap.

I tried using someone's FTDI library, which used a synchronous mode to drive WS2812B.  It might have worked, but I verified with a scope that the clocking took the timing out of spec.  I see no reason to expect I2C would be any better.
No such problem with the Pico's PIO, because it's bitbanging by HW coprocessor.  I'm thinking I could even emulate a WS2811 entirely, which would allow a Pico module to do some very interesting things without additional hardware.

Ok, I'm done geeking out.  Ciao.