wled/WLED #1710: Ws2815 power on problem.

ddijanic, 2021-01-30:
I have esp 07 runing WLED_0.11.1_ESP8266_1M.bin whit WS2815.
using PC power supply , 3,3 v for esp and 12v for WS2815 12v.

when i turn the power supply, led strip goes 1,2, Led white,3 led red, and esp no wifi-ap ... esp led is on, but no wifi.
When i disconnect the strip. power off and power on again esp wifi AP is working, then when i connect the strip is working.

tribe disconnecting GND from strip and power on, not working
I tried disconnecting 12+ from strip, power on, not working.
it only power on when GND and +12V are not connected to strip
I tried putin capacitor to strip and to esp, not working
I tried 12v power supply , and dc-dc for esp, not working.

i putt double switch on strip for gnd and + , switch set to OFF, power on power supply, then switch ON, and then is working .

ws2815 gnd to gnd (same gnd for esp ), 12+, to power 12+,
and D0 do esp..  
I tried BO and Do to esp, not working, only if switch is OFF, when powering and after switch ON

i didn't tried BO to GND maybe i need that dont know.

DivanX10, 2021-02-01:
вы с какой стороны подключаете диоды? там есть указательная стрелка, если подключить против стрелки, то будет то, что вы и описали. Для теста  я подключал диоды с обратной стороны, против стрелочки и как раз у меня происходит то, что вы и описываете.

which side do you connect the diodes from? there is a pointer, if you connect the opposite direction of the arrow, then will be what you described. For the test, I connected the diodes on the back side, against the arrow, and just what you describe is happening to me.
[image]

и не понятно как вы их подключили. Вы так их подключили, как на схеме?
and it is not clear how you connected them. Did you connect them as shown in the diagram?
[image]

ddijanic, 2021-02-01:
Hay .
I arrow direction was good. But i will check again. 
Yes Asus diagram connect, a tried dana to DI and BI. 
Thet i when back data only to DI.

Thanks for replay...

Apapra, 2021-02-01:
Not all WS2815 works with 3,3V data channel. Maybe you need a level shifter 3,3V → 5V.
But ESP should work anyway.

ddijanic, 2021-02-11:
[image]

Wireing Is good, its needs Level shifter to 5v probably.dont have it home to try it. Mesto order and whait 30 data. Will stay laki this then.

closed 2021-06-28 by stale[bot]

filzek, 2023-03-15:
Hello Fellows,

The WS2815 has a interconnect Backup data with the Data Input, the Led Strip does need to redirect the backup line to the input line on each led, this solve the problem when the LED got burned and it need to forward the DATA to the next led, so, it will work always when the Data Out is missing the WS2815 led will read the data from the BI line.

So the first led will always have the BI connected to the GROUND!

[image]