wled/WLED #1712: Segments - Starting LED, Number of LEDs, seems wrong

Eddiiie, 2021-01-31:
**Describe the bug**
I looked but did not see anything about this, so maybe it is just me being anal.     I have 2 segments of 60 LEDs each.
UI cosmetic bug I'd say.

Web UI, Segment 0.    Starting LED 0,  Ending LED 59  ?    The result of this says 59 LEDS,  not 60.  
So I make it Starting LED 0, Ending LED 60.   The result says 60 LEDs.  (not 61).

Segment 1, Starting LED says 60.  (not 61),  ending LED is 120.   The result of this says 59 LEDs, not 60.
So I make it Starting LED 61, ending LED 120.  The result says this is 59 LEDs.

It seems to work OK like this.    But seems odd to have Starting LED at 0  and ending LED 120.  One segment with 60 LEDs and one with 59.

[image]

**To Reproduce**
Just Edit first segment (0), set ending LED to 59?  It should say 60 LEDs.
Create a second segment (1), starting LED should be 60? 61? I'm confused now.

**Expected behavior**
I'd expect starting LED to be 1; not 0, even though it is probably address 0.  
I'd expect ending LED to be the last LED (120).  
I'd be OK if it starts at 0.   The UI should then count 0 as a 1 and ending LED should be -1 (59 in my case)
I want 2 segments of 60 LEDs each.

**WLED version**
 - Board: Wemos D1 mini
 - 0.11.1, [build 2101130]
 - Format [self-compiled]

**Additional context**
I'm new here... :p

Thank you for your help!

giddyhup, 2021-03-23:
I am observing the same. Additionally, when creating two consecutive segments (0-20 and 21-40) one LED remains uncontrolled, this literally doesn't add up. 

v0.12.0-b0 "Hikari" - self compiled
Board: Wemos D1 mini

Legsmaniac, 2021-03-23:
> when creating two consecutive segments (0-20 and 21-40) one LED remains uncontrolled, this literally doesn't add up.

This should be 0 to 20 and then 20 to 40.  Yours will be why you have a missed LED.

Legsmaniac, 2021-03-23:
> I looked but did not see anything about this

https://github.com/Aircoookie/WLED/wiki/Segments

giddyhup, 2021-03-23:
> This should be 0 to 20 and then 20 to 40. Yours will be why you have a missed LED.

My bad, I skipped over this part in the documentation: "The Stop LED is not included in the Segment." Thanks for pointing it out.

closed 2021-07-21 by Aircoookie