wled/WLED #1716: White does not show up on PEEK

Skiidlive, 2021-02-02:
White does not show up on PEEK viewer when using RGBW and "accurate" is selected.  All effects and colors work correct on the led's but anything white no longer shows in the PEEK viewer.  If I change the white setting to say "both" it works.  Using SK6812 strips.

Aircoookie, 2021-02-04:
Thank you for the report!
This is an issue with conversion from RGBW back to RGB for display in the Peek bar. I have made a patch in the dev branch so that at least the RGB component of the color is correctly displayed. It will be merged to master soon and be part of the next release :)

Skiidlive, 2021-02-04:
Awesome. Thank you.  Amazing work you are doing.

closed 2021-02-04 by Skiidlive

ChuckMash, 2022-01-26:
Seems like this is still happening, I'm not seeing white channel data in peek / websockets live view data.

commit 2c5eba33 (2022-01-26): Added white channel to Peek (closes #1716)