wled/WLED #1717: WLED Starlight

LeonvG, 2021-02-02:
Hey you all. I want to create a simple Starlight with WLED and two motion sensors but I really don't know how to do it. Im using Home assistant so it would also be really awesome if the two motion sensor were imported into home assistant. The Starlight should light up in like a wave motion in the direction the object moves and stay on for like a minute or so. Do anybody know how I can do this all ?

asterixonline, 2021-02-02:
You can use the motion sensors in home assistant to trigger a service to call wled with a preset / effect.

https://community.home-assistant.io/t/how-to-turn-on-wled-via-automation/147187/14

If you find the right effect or make a preset, and you time it well, you can stop it after xx seconds (when it's on the end of the stairs).

Maybe this brings you to the right direction?

LeonvG, 2021-02-02:
Yeah it is the right direction but I don't have motion sensors in my hallway. So I wanted them to connect to the d1 mini wich is running WLED. Furthermore it would be awesome to integrate these motion sensors into home assistant.

closed 2021-06-18 by stale[bot]