wled/WLED #1718: Editor (/edit) is partially broken when using LittleFS

embedded-creations, 2021-02-02:
I'm describing it as a feature request to fix not a bug, as it's working well enough for WLED's needs as is.

Browsing files in folders (among other things) is broken.  This is a known issue:

https://github.com/me-no-dev/ESPAsyncWebServer/issues/879
 
The fix seems to be pulling some of the changes in @lorol's fork into the WLED fork.

Aircoookie, 2021-02-04:
Yes, I was aware of the currently non-functional support for folders as well. To support more advanced features in FS (sequences, custom palettes, maybe even dynamic FX at some point), folder support in the `/edit` page would be great.

An additional sticking point is the way the file list is implemented. It makes heavy use of arduino String concatenation which is prone to creating heap fragmentation. I try to avoid them in WLED as much as possible, instead using oldschool c strings, but some dependencies, especially ESPAsyncWebServer use them.

embedded-creations, 2021-10-30:
This is being discussed in #1592