wled/WLED #1720: invalid json in presets.json

SebastianHe, 2021-02-04:
When you add preset add an other id as 1, then it's add an additonal "," at the start. 
So that the json file is not valid. 

presets.json

`{                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      ,"13":{"n":"test","on":true,"bri":255,"transition":1,"mainseg":0,"seg":[{"id":0,"start":0,"stop":233,"grp":1,"spc":0,"on":true,"bri":125,"col":[[233,0,220],[211,83,254],[111,187,56]],"fx":29,"sx":136,"ix":74,"pal":0,"sel":true,"rev":true,"mi":false},{"id":1,"start":0,"stop":1,"grp":1,"spc":0,"on":true,"bri":255,"col":[[123,41,0],[0,122,254],[254,203,61]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":2,"start":0,"stop":1,"grp":1,"spc":0,"on":true,"bri":255,"col":[[97,23,123],[254,247,107],[254,55,44]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":3,"start":0,"stop":1,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,0,0],[177,140,254],[195,209,22]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":4,"start":0,"stop":1,"grp":1,"spc":0,"on":true,"bri":255,"col":[[254,251,64],[254,251,64],[254,251,64]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}}`

In this case i added a preset as id 13 - and i had no preset as id 1.

**WLED version**
 - Board: ESP32
 - Version version 0.11.1
 - Format Binary

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

Aircoookie, 2021-02-04:
Hi and thank you for the report!
When `presets.json` is created, it normally has a faux preset 0 at the start to prevent exactly this issue (the initial contents of the file are `{"0":{}}`). For some reason it is not present in your file. I will try to reproduce the issue. Did it happen on a fresh install or did you make some other presets and then deleted them or edited the preset?

SebastianHe, 2021-02-04:
It's simple to reproduce:

delete all presets, add a new preset (via the webinterface) at id 13.

Aircoookie, 2021-02-04:
It preserves the `0` preset for me as it should be... Both deleting all presets and deleting the entire presets.json file work.
```json
{"0":{},"13":{"n":"test 13 2","on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":25,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,160,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}}
```

Must be more to it. Will continue to investigate :)

SebastianHe, 2021-02-04:
ah ok, 

then it must be my mistake. 

I cleared the file with "{}" and then added the entry.

closed 2021-06-16 by stale[bot]