wled/WLED #1723: Randomly first LED turns on Green when the WS2812B is turned off

bachoo786, 2021-02-05:
**Describe the bug**
I have a ws2812b 5m long with 60leds in each metre i.e. 300 leds in total. I have connected a 470ohm resistor between the data line and esp8266. I also have 2x 1000uf capacitors connected i.e. one of them is between the positive and negative 5V supply that powers the first 1.5m of the led strip.

The second 1000uf capacitor is connected between positive and negative 5V supply that power the rest of the 3.5m of the led strip.

Randomly when the led is turned off I see the first led on and is green in color. I had this issue even before I connected the resistor and capacitors.

**To Reproduce**
N/A its a random behavior

**Expected behavior**
When the led strip is turned off I expect all the leds to remain off and not any turning on by itself and being green in color. 

**WLED version**
 - Board: Wemos D1 mini
 - Version: 0.11.1
 - Format: Binary

sansillusion, 2021-02-05:
This is sign of static in the data line, you could try twisting rhe data
line with ground or use a shielded cable for data. Or you could use the
relay mod to turn off power to the leds when off.

bachoo786, 2021-02-05:
Hi @sansillusion. Thanks for the reply.
1. Are you saying to connect the data line to ground and twisting both the cables together?
2. Shielded data cable. That is a good option.
3. Relay mod? I did not understand the last option. Can you please explain? 
Thanks again.

sansillusion, 2021-02-06:
1 not connecting both wires just twisting the wires together like a network cable (twisted pair).

3 there is a wled usermod that can be used to control a relay using a gpio of the esp.

bachoo786, 2021-02-06:
I think I will look into the 3rd option but i am not sure still
 I have buried the data and gnd cables.

toto79, 2021-02-07:
> 3 there is a wled usermod that can be used to control a relay using a gpio of the esp.

it isn't a usermod is it a built in feature see Line 32 in [https://github.com/Aircoookie/WLED/blob/94941a773292aaab3ac121bb5a597699a626ea80/wled00/NpbWrapper.h#L32](url)
in combination with Line 40.

if you use the standard image for wemos d2 mini, then you should have 3.3V on GPIO12=GPIO6 if led is wled switched on, and 0V if Switch in wled is off. You only have to add a relais in line to switch of VCC/5V 

see also here for some schematics: [https://wled.discourse.group/t/strong-light-flash-when-powerup-ledstrips/145](url)

bachoo786, 2021-02-07:
> it isn't a usermod is it a built in feature see Line 32 in [https://github.com/Aircoookie/WLED/blob/94941a773292aaab3ac121bb5a597699a626ea80/wled00/NpbWrapper.h#L32](url)
> in combination with Line [...]

My wled is powered from the 5v adapter not the wemos mini. Can I still use option 3?

Also would my wled strip stop working if I don't do anything?  I occasionally restart the wled strip when I see my 1st led turned on.

toto79, 2021-02-07:
the relais can contraoll/switch whatever you want. the green color is more an optucal issue, you shouldn't need to restart your wled to get it work again.

bachoo786, 2021-02-08:
@toto79  thanks :)

if I was to ignore this first LED randomly turning on, would it have any effect on the rest of the LEDs? I mean if I did not resolve the issue and the 3 options that @sansillusion gave me what is the worse that could happen?

1. The reason why I say this is because my data and ground cables have been buried in a wall and I cannot access them to twist them. I may be only be able to access part of them like 30cm when they connect to the esp8266 but the rest of the length is all buried and I suppose twisting 30cm of the length wouldnt help alot ?

2. I cannot use a shielded cable as the same problem mentioned above I have buried my data and ground cables in the wall so I cannot access them.

3. @toto79  the links you provided in your answer above do not contain any information I think its deleted? Again using a relay to turn the LED is it going to help? because I have to keep an eye out for the first LED when it turns on and as soon as it does I should turn it off using the relay. This is not practical compared to the first 2 options?

sansillusion, 2021-02-08:
The relay option will turn off the leds power when they are turned off in
wled. And turn it back on when wled is switched back to active.

sansillusion, 2021-02-08:
Also since you are using long data wire you may (just mabe) get better
signal if you use a level shifter to change the signal from 3.3V to 5V

bachoo786, 2021-02-08:
@sansillusion the relay option is not smart as I have to monitor the led myself. 

I will add a level shifter and see how i get on thanks.

bachoo786, 2021-02-08:
@toto79 sorry I think I don't understand when you say relay. Do you mean adding a physical relay to the esp8266?

Or is it something software based?

toto79, 2021-02-09:
i meant a physical one -like it is described in the discourse link. :)

hannesbe, 2021-02-21:
Try reverting to 0.10.2, helps with a somewhat similar issue:

https://github.com/Aircoookie/WLED/issues/1693

closed 2021-06-30 by stale[bot]