wled/WLED #1727: NTP time sync get random values

gaelbenoit, 2021-02-06:
**Describe the bug**
Since I have upgraded my WLED builds to 2101130, the ntp client give random values. After a reboot the time is correctly synced, but 24h later it's absolutly random:

1st: **ESP32 DevBoard**
NTP: **local ntp server**
**Current local time is 2061-9-8, 7:13:27.**
board = esp32dev
platform = espressif32@1.12.4
build_flags = ${common.build_flags_esp32} 

2nd: **ESP32 D1 MINI**
NTP: **pool.ntp.org**
**Current local time is 2006-5-16, 9:44:09.**
board = esp32dev
build_flags = ${common.build_flags_esp32} 
platform = espressif32@2.0

3rd: **ESP32 D1 MINI**
NTP: **0.wled.pool.ntp.org**
**Current local time is 2061-9-8, 7:16:31.**
board = esp32dev
build_flags = ${common.build_flags_esp32} 
platform = espressif32@2.0

4th: **ESP8266 D1 MINI**
NTP: **pool.ntp.org**
**Current local time is 2017-6-24, 20:17:44.**
board = d1_mini
platform = ${common.platform_wled_default}
build_flags = ${common.build_flags_esp8266}

**To Reproduce**
I just downloaded the release 2101130 and build it with PlatformIO and my historical platformio_override.ini file

**Expected behavior**
The correct time from NTP

**WLED version**
 - Board: ESP8266 D1 MINI, ESP32 Dev Board, ESP32 D1 MINI
 - Version 0.11.1 2101130
 - Format self-compiled

closed 2021-06-16 by stale[bot]

lodesmets, 2023-11-11:
Do you still have the problem? Because I have it too with v0.14.0

Only very infrequently

jannic, 2023-11-19:
I also have this issue, with v0.14.0 on an esp8266. Usually happens after several days of uptime.

softhack007, 2023-11-20:
@jannic @lodesmets please try latest source code from 0_14_1 (PR https://github.com/Aircoookie/WLED/pull/3536).

More details are here: https://github.com/Aircoookie/WLED/issues/3515#issuecomment-1819754367

lodesmets, 2023-11-21:
I compiled it myself with the USERMOD_SSDR. For now it seems to work, but the problem only came after a long uptime. So let's hope this works too