wled/WLED #1728: Only first 3 LEDs light up, no control

senex44, 2021-02-06:
Hi! 
I have an ESP8266 (exactly [this ](https://www.az-delivery.de/products/copy-of-nodemcu-lua-amica-v2-modul-mit-esp8266-12e?variant=36247507474) here) 
and a BTF-LIGHTING WS2812B ([amzn](https://www.amazon.de/gp/product/B088BRY2SH/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1)). 
I've flashed WLED_0.11.1_ESP8266_ledpin4.bin to the ESP and can see the WLED-AP. I've included it in my network and can access the WLED web page. 
**Nevertheless, only the first 3 LEDs light up, first one blue, second green, third red. There is no control possible,** they just light up when they get power from the ESP. 
Data cable doesn't change this (plugged in/plugged off). 
The ESP'S blue light lights up, when the data cable is connected to D4, not when it is connected to D2. 
I've tested the data cable at both ends of the light strip, but for normal I am working in flow direction of the small arrows on the strip.
[image]

asb2106, 2021-02-07:
GPIO pin 4 is actually pin D2 on the nodemcu you got there :) Move your green data wire there from D4 to D2 and you should light up.  you wont have a lot of power for the lights going through the usb like that, ive never tried.  get some power injected and you are good to go

senex44, 2021-02-07:
Unfortunately, this does not make any difference with a separate power unit. But I found a LED ring, and it's directly working smoothly. So I guess the LED stripe is somewhat broken or so. Sorry for that and thanks for the advice!

closed 2021-02-07 by senex44