wled/WLED #1731: "Set current preset cycle setting as boot default" is not working

gsieben, 2021-02-07:
**Describe the bug**
I created a cycle which works very well. But when I restart the device the "Set current preset cycle setting as boot default" is ignored. 

**To Reproduce**
1. Create some presents. 
2. Do your cycle settings. 
3. Check the "preset cycle". (If it checked - uncheck - do your settings again - check. "Thats another bug)
4. "Set current preset cycle setting as boot default" in Config.
5. Repower -> Not working. 

**Expected behavior**
Preset cycle on startup. 

**WLED version**
 - Wemos D1, ESP32 dev]
 - Version [newest]

THX Gabriel :-D

Model-Man, 2021-02-13:
Example.

Lets say you have 40 presets. You wish to run 23-38 at startup. Set "Apply preset 23 at boot" with the "Preset cycle" box checked at "First preset 23" and "Last preset 38". This works for me.

Questman01, 2021-02-21:
For me the same. I does not work. After restarting WLED the preset cycle settings are back to default.

closed 2021-06-30 by stale[bot]

ishdemon, 2021-11-06:
Issue still persists in 0.12.0😬

blazoncek, 2021-11-06:
0.12 is from April. Use 0.13-b4.