wled/WLED #1733: Errors while compiling with VSC

DSRocker, 2021-02-08:
I get always errors after compiling:

The terminal process "C:\Users\DS\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: C:\Users\DS\.platformio\penv\Scripts\pio.exe run <

Processing d1_mini (board: d1_mini; platform: espressif8266@2.6.2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.7 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Library Manager: Installing git+https://github.com/lorol/LITTLEFS.git
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden:
  File "C:\users\ds\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\DS\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\DS\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\ds\.platformio\platforms\espressif8266@2.6.2\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\DS\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\users\ds\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\DS\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\users\ds\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\DS\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\users\ds\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\users\ds\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\ds\.platformio\penv\lib\site-packages\platformio\proc.py", line 117:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\DS\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\DS\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)

denisTenis1, 2021-02-17:
Did you find a fix?

DSRocker, 2021-02-17:
Yes. Thanks

denisTenis1, 2021-02-17:
I have the same problem, could you tell me how did you fix it?

denisTenis1, 2021-02-17:
Can you tell me how did you fix it?

srg74, 2021-02-18:
@denisTenis1 For help you can ask at WLED Discord channel

DSRocker, 2021-02-18:
Don’t change the GPIO-Outputs. It’s not as easy to find the right files to change something –without reading all the documentations-

The 2. Problem was: I used a 10k resistor from GPIO2 to LED DataIn. It’s to much. 200 Ohm is ok.

closed 2021-06-28 by stale[bot]