wled/WLED #1736: Issue with ethernet support with wesp32

jimcortez, 2021-02-11:
**Describe the bug**
I used the `esp32-poe` environment to build and upload to my new wesp32 device. I have it connected via a known-working ethernet port. WLED comes up and starts an AP, however it does not seem to use the ethernet port to get an IP. I tried setting the ethernet type to poe again, but no luck.

I enabled debug logging by adding `${common.debug_flags} ` to the esp32-poe target, but nothing there seems interesting.

**To Reproduce**
Compile latest master (commit `f7114fc2aa436b026075e26b480dd0e883b713c0`) with esp32-poe target, load onto wesp32 device. See that it does not get an IP, but instead starts an access point

**Expected behavior**
Compile with ethernet support, device should auto-negotiate an IP with DHCP, then be accessible via that IP.

**WLED version**
 - Board: wesp32 https://wesp32.com/
 - Version f7114fc2aa436b026075e26b480dd0e883b713c0
 - Format self-compiled

**Additional context**
None, this is a new device. Happy to enable any additional logging! Looks like the support should have been available since #1583 

Thank you for the help! Been pretty happy with this project with some existing esp8266 devices. Keep up the awesome work!

k7bbr, 2021-02-11:
OK. I made the changes and compiled a binary. I don't have a WESP32 board to test. You can get the binary here:

https://github.com/k7bbr/WLED/releases/tag/0.11.1_beta

Let me know if it works. If it does I'll push the changes to the master branch. Thanks.

Also, a note: once you change the wifi settings to enable your board, plug the Ethernet cable in and then cycle the power on your board. This way it will attempt a dhcp request via the Ethernet port prior and not re-create an access point.
-B

lonestriker, 2021-02-11:
I don't have a WESP32 board to test with, but your changes look good.  The one bit that I didn't have to worry about was actually the board type when adding the previous WT32-ETH01 support. I just used either the esp32dev or esp32-poe boards, either of which worked for the ESP32-POE and WT32-ETH01 boards.

A non-technical assessment: I'm not sure what the market is for this WESP32 board though.  $50 for an ESP32 POE board is 2x the price of the other ESP32-POE board and 10x the cost of the non-POE WT32-ETH01 board.  $50 for a microcontroller+POE is getting into SBC prices for Raspberry Pi and other much more full-featured devices.

jimcortez, 2021-02-12:
@k7bbr I appreciate the fast help! Unfortunately it did not seem to work. I doesn't seem to negotiate an IP over ethernet and does not pop up a WLED-AP ssid. The serial monitor does connect, but I don't see any output. Is there some other information I can provide?

k7bbr, 2021-02-12:
Hi Jim, Sorry that didn't work. I tried it on my WT32_ETH01 board and it popped up the WLED-AP and when I went to wifi settings I was able to select my board and see the new drop down for the WESP32. 

It almost sounds like having the serial monitor up might be related? Can you just connect 5V and GND to the board and see if it will bring up the access point? Is it still in bootloader mode perhaps? Sorry I don't have a board here to test with.

jimcortez, 2021-02-14:
I have played around with your build and was able to get it to work by changing https://github.com/k7bbr/WLED/blob/a09f64aee5da2b506673d2b92edf182bd2650bd9/wled00/wled.h#L208

from 
```
WLED_GLOBAL int ethernetType _INIT(WLED_ETH_ESP32_POE);
```
to
```
WLED_GLOBAL int ethernetType _INIT(WLED_ETH_WESP32);
```

I suspect that one or more pins that are being configured prevent the system from booting up properly, which doesn't allow the AP to start, which means I cannot select the proper board in the menu system.

I will admit that this is a maybe, I don't know this codebase very well and don't know what the right way to configure the build time variables should be here.

lonestriker, 2021-02-15:
@jimcortez @k7bbr That makes sense. I originally put in a "none" option in that field so that it would not initialize any ethernet device and force users to login to the Wi-Fi AP to set it themselves. If we have a default configured, it may try to initialize things incorrectly for the wrong board.

k7bbr, 2021-02-15:
@jimcortez @lonestriker Great work to figure that out Jim. Glad you got it working.

Now that we have 3 supported boards, I wonder if we should initialize it to none by default and then let the user turn on the ethernet appropriately for their board? We could also make it a compile time option in Platformio. Maybe do both: Initialize to none and make a compile time option to set the default to a particular board if they wanted to have it set.

Aircoookie, 2021-02-18:
@k7bbr Yeah, I believe initializing it to none by default is the best way for a general ethernet build. We can also make a macro however to set a specific board as default at compile time!

jimcortez, 2021-02-19:
@k7bbr @Aircoookie Only issue is that setting this menu item to the wrong value would bork the setup on wesp32. Setting it to the wrong value prevents the network from coming up. This means I have to re-flash to get it working.

I should say I am ok with this, adding custom components anyway. However, could be very confusing for folks in the future!

k7bbr, 2021-02-19:
@Aircoookie Ok. Sounds good. I made changes to make None the default for the general ethernet build in addition to adding the WESP32 support. I also made a macro/build flag to set a default ethernet board if one is desired at compile. I also changed the env in platformio.ini to esp_eth as we discussed in another thread. Look for a PR once I clean things up. I wasn't exactly sure the best nomenclature to use for the compile time macro, but from my testing, what I had in there works just fine. It defaults to None, and then if a different board is defined as default at compile time, it uses that board as default instead.

@jimcortez Very good points about the WESP32. That could be an issue if someone fat-fingers that setting. I suppose we could make another env for just the WESP32, and hide the other board options except for None in the WESP32 board env. Only problem is that gets away from having just a general eth board type.

closed 2021-06-28 by stale[bot]

kgreene1983, 2022-01-31:
@k7bbr 

Have found a new board on Aliexpress and I was hoping to see if you could work out how to get it going? I can't work out the pins.. If you think it might work I'll order one to play with it as they're fairly cheap.

https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE

Vertikar, 2023-04-25:
In case anyone stumbles across this and wonders if it was fixed, looks like it was under: https://github.com/Aircoookie/WLED/pull/1764

gon0, 2025-02-12:
I have a WESP32 and installed WLED 0.15.0 (and 0.14.4 for testing) with the "Ethernet" option selected via web installer.

I have changed the LED GPIO to GPIO3, so that it does not interfere with the GPIO16 needed for Ethernet.
After that I selected the WESP32 board under "WiFi settings".

At the WESP32 board only the green ethernet LED is blinking.  The yellow LED is off. I cannot find the board via Ethernet. I have not entered WiFi SSID or Password and left the IP at 0.0.0.0, so the DHCP server should provide an IP.

Are there any steps I might have missed?