wled/WLED #1739: Hardware support for ESP32 D1 MINI

mgcrea, 2021-02-12:
I recently bought a pack of 3 [ESP32 D1 MINI](https://www.az-delivery.de/en/products/esp32-d1-mini) that I intended to use with WLED.

Unfortunately it does not work out of the box, tried both `default` and `gpio16` builds.

Did build and flash locally trying many supposedly available pins with no luck.

Tried with and without a logic level shifter.

Uncommented `#define USE_WS2801` (not sure if required but I'm driving ne-pixels).

Any ideas of things I could try?

k7bbr, 2021-02-12:
I've used the ESP32 minis. Try compiling with the `[env:wemos_shield_esp32]` build.

mgcrea, 2021-02-12:
Just realised that I forgot to toggle the RGBW setting -_-, it's working now, thanks for the help anyway!

closed 2021-02-12 by mgcrea